isaac-sim/IsaacLab #2055: [Question] Issue with MultiAssetSpawnerCfg in RL training

luju97, 2025-03-10:
### Question

Hi,

I'm trying to train my robotic arm to pick up different types of objects, starting from the `FrankaCubeLiftEnvCfg` example.

I attempted to use `MultiAssetSpawnerCfg` to spawn a different object in each environment, but I noticed that the same object is spawned across all environments. Why is it?

Thanks in advance!

virgilio96upm, 2025-03-10:
Hi, did you set `replicate_physics=False` in your env? (https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.scene.html) If you are using the manager based approach add         `self.scene.replicate_physics= False `inside of your `FrankaCubeLiftEnvCfg`. I hope this solves your issue!

luju97, 2025-03-11:
Hi, it solved the issue:) Thank you!