isaac-sim/IsaacLab #2062: [Question] Question title

MiangChen, 2025-03-12:
### Question

Ubuntu 22.04 6.8  + kernel 6.8.0
NVIDIA driver 535.216.01
CUDA 12.2
IsaacSim: 4.2 installed from .zip

when I try to run `./isaacsim.sh`
the probelm occurs: 
```
2025-03-12 14:26:34 [423ms] [Error] [gpu.foundation.plugin] No device could be created. Some known system issues:
- The driver is not installed properly and requires a clean re-install.
- Your GPUs do not support RayTracing: DXR or Vulkan ray_tracing, or hardware is excluded due to performance.
- The driver cannot enumerate any GPU: driver, display, TCC mode or a docker issue. For Vulkan, test it with Vulkaninfo tool from Vulkan SDK, instead of nvidia-smi.
- For Ubuntu, it requires server-xorg-core 1.20.7+ and a display to work without --no-window.
- For Linux dockers, the setup is not complete. Install the latest driver, xServer and NVIDIA container runtime.

|---------------------------------------------------------------------------------------------|
| Driver Version: 0             | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|=============================================================================================|

```

I also take this for (reference)[https://forums.developer.nvidia.com/t/nvidia-driver-nvidia-linux-x86-64-535-129-03-run-installation-failed-on-ubuntu-22-04/304564/4], the recommand nvidia driver version is 535.216.01 as mine. It is strange. 
did anyone meet the same problem?

[image]

RandomOakForest, 2025-03-13:
Thank you for posting this. It is likely a driver issue. Could you try 535.129.03, and follow the installation steps for Isaac Lab 2.0.2 and Isaac Sim 4.5 [in this doc](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html)?

MiangChen, 2025-03-14:
> Thank you for posting this. It is likely a driver issue. Could you try 535.129.03, and follow the installation steps for Isaac Lab 2.0.2 and Isaac Sim 4.5 [in this doc](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html)?

hi, I have resolved this problem. The main reason is that I added a parameter "-no-opengl-files" the first time. 
so it's necessary to choose **install opengl** when install the NVIDIA driver, althougt there are many blogs saying opengl is useless...