isaac-sim/IsaacLab #2064: [Proposal] Add  method to receive seperate contact and friction force from the force sensor

ParlitsisG, 2025-03-12:
### Proposal

Add a seperate contact and friction force from the force sensor instead of their sum.

### Motivation

There are certain applications that require the force between objects minus the friction force. Such as reward sytems for dual arm systems grasping objects requiring the contact forces of end effectors to be opposing each other in direction.

### Alternatives

I am currently trying to get this information from PhysX via rigid_contact_view.get_friction_data(dt) but is requires additional work to pull this information. (I will update if I manage). But generally a simple way to pull this information would be Ideal.

### Additional context

For example currently in a dual manipulation set up using two modified franka emika arms with flat end effectors I want to have the forces between the two end effectors to be opposing in order for the robot arms to grasp the box firmly. Yet if I implement this in a reward system if the robot lifts by pushing equally from both sides the friction forces will provide a net force pointing at different directions.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)

RandomOakForest, 2025-03-13:
Thank you for posting this proposal. The team will review it.

virgilio96upm, 2025-03-14:
Hi, I don't know if this is helpful, but I did a first implementation of the PhysX `get_contact_data` function from omni.physics.impl.api to try to get the detailed contact data between sensors and filter prims per patch. The details of the implementation and an example using different cubes can be found [here](https://github.com/isaac-sim/IsaacLab/discussions/2026)

AntoineRichard, 2026-08-27:
Thanks @ParlitsisG for the proposal — this is now in.

`ContactSensorCfg.track_friction_forces=True` populates `ContactSensorData.friction_forces_w`, the summed friction force per filtered shape, alongside `net_forces_w` / `force_matrix_w` (https://github.com/isaac-sim/IsaacLab/pull/3563). Two caveats: it needs `filter_prim_paths_expr` configured, and it's PhysX-only for now — the OVPhysX contact sensor raises `NotImplementedError` if you enable it, and the Newton one warns and forces the flag off. Closing as implemented; please reopen if the per-contact breakdown you need still isn't reachable.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard