isaac-sim/IsaacLab #2067: [Question] Behavior of joints not actuated by policy output

T-K-233, 2025-03-13:
### Question

In Manager based environments, we can configure the action space of the policy using the `ActionsCfg`. In `JointPositionActionCfg`, we can use the `joint_names` term to specify which actuators we want to control, some thing like:

```python
# env_cfg.py

@configclass
class ActionsCfg:
    """Action specifications for the MDP."""

    joint_pos = mdp.JointPositionActionCfg(
        joint_names=[<list of joint names we want to drive>],
        preserve_order=True,
        use_default_offset=True,
    )
```

My question is that for the actuators that are not specified in the list, what is the desired behavior?

From my test, it seems that those joints will be initialized to their default init positions (default offsets), but then will be driven to the zero position defined in the USD files. Is this the intended behavior?

In my opinion, the joints should stay at their default init positions as if the policy is outputting "target_pos = 0" to the actuators. This is useful in cases where we only want to control part of the robot. Example would be we only want to control the arms / legs of a humaonid robot.

Mayankm96, 2025-03-13:
> From my test, it seems that those joints will be initialized to their default init positions (default offsets), but then will be driven to the zero position defined in the USD files. Is this the intended behavior?

Yes, that's the intended behavior right now. We let the user specify the targets for the joints and not go for a default behavior as this depends on the use-case. Right now the joint target tensor `joint_position_target_sim` is initialized at zero so if it is not getting set by somewhere (for instance actuators), the values are zero.

> In my opinion, the joints should stay at their default init positions as if the policy is outputting "target_pos = 0" to the actuators. This is useful in cases where we only want to control part of the robot. Example would be we only want to control the arms / legs of a humaonid robot.

For joints that you don't want to control, you can set their stiffness/damping gains to zero. This would make PhysX apply no efforts on them. However, if you want them to stay at some default configuration (i.e. the joint position targets are the default joint positions), you need to make a dummy joint action term that sets the joint position targets to the default joint state. Alternately, you can make an event term that sets the joint position targets to the desired values for the unactuated joints.

This is what I do currently:

```python
from __future__ import annotations

import torch
from typing import TYPE_CHECKING

from omni.isaac.lab.assets import Articulation
from omni.isaac.lab.managers import SceneEntityCfg
import omni.isaac.lab.utils.math as math_utils

if TYPE_CHECKING:
    from omni.isaac.lab.envs import ManagerBasedEnv

def randomize_joints_targets_within_limits_range(
    env: ManagerBasedEnv, env_ids: torch.Tensor | None, asset_cfg: SceneEntityCfg
):
    """Randomize an articulation's joint targets to a random position in the joint limits range."""
    # extract the used quantities (to enable type-hinting)
    asset: Articulation = env.scene[asset_cfg.name]

    # resolve environment ids
    if env_ids is None:
        env_ids = torch.arange(env.scene.num_envs, device=asset.device)

    # resolve joint indices
    if asset_cfg.joint_ids == slice(None):
        joint_ids = slice(None)  # for optimization purposes
        num_joints = asset.num_joints
    else:
        joint_ids = torch.tensor(asset_cfg.joint_ids, dtype=torch.int, device=asset.device)
        num_joints = len(joint_ids)

    # randomize the joint targets
    joint_pos_limits = asset.data.soft_joint_pos_limits[env_ids, :]
    joint_pos_limits = joint_pos_limits[:, joint_ids]

    # sample the joint targets
    shape = (len(env_ids), num_joints)
    joint_pos_targets = math_utils.sample_uniform(
        joint_pos_limits[..., 0], joint_pos_limits[..., 1], shape, device=asset.device
    )

    # set the joint targets
    asset.set_joint_position_target(joint_pos_targets, joint_ids=joint_ids, env_ids=env_ids)
```

T-K-233, 2025-03-20:
I see. Thanks for the explanation and example code!

closed 2025-03-20 by T-K-233