isaac-sim/IsaacLab #2068: [Bug Report] Training Script Hanging

alextong1010, 2025-03-13:
### Hanging
I'm trying to resume a run and when I run my training script, it hangs after

```
[INFO]: Starting the simulation. This may take a few seconds. Please wait...```

### Steps to reproduce

Create a bash script like this and run it:
```SCRIPT_PATH="scripts/reinforcement_learning/rsl_rl/train.py"
TASK="Isaac-Velocity-Flat-Unitree-Go2-v0"
NUM_ENVS="4"
MAX_ITERATIONS="10000"
SEED="100"
VIDEO_LENGTH="200" # in steps
VIDEO_INTERVAL="2000" # in steps

# Checkpoint parameters
RESUME="True"
LOAD_RUN="2025-03-10_13-55-55"  # Replace with your actual run folder name
LOAD_CHECKPOINT="model_9999.pt"  # Replace with the iteration number to load

# Use this to toggle which cuda device gets used for training
export CUDA_VISIBLE_DEVICES=0
export HYDRA_FULL_ERROR=1

# Run the command with specified arguments
./isaaclab.sh -p "$SCRIPT_PATH" --task "$TASK" --num_envs "$NUM_ENVS" --seed "$SEED" \
  --max_iterations "$MAX_ITERATIONS" --video --video_length "$VIDEO_LENGTH" \
  --video_interval "$VIDEO_INTERVAL" --resume "$RESUME" --load_run "$LOAD_RUN" \
  --checkpoint "$LOAD_CHECKPOINT"
```

### My Result

```
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.locomotion.velocity.config.go2.flat_env_cfg:UnitreeGo2FlatEnvCfg
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.locomotion.velocity.config.go2.agents.rsl_rl_ppo_cfg:UnitreeGo2FlatPPORunnerCfg
[INFO] Logging experiment in directory: /home/jordan/alex/IsaacLab/logs/rsl_rl/unitree_go2_flat
Exact experiment name requested from command line: 2025-03-12_22-12-32
Setting seed: 100
[2025-03-12 22:12:32,584][ogn_registration][INFO] - Looking for Python nodes to register in omni.physx.fabric-106.5.7
[2025-03-12 22:12:32,584][ogn_registration][INFO] -  -> Registered nodes from module omni.physxfabric at /home/jordan/anaconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.fabric
[2025-03-12 22:12:32,584][ogn_registration][INFO] - Registering nodes in /home/jordan/anaconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.fabric imported as omni.physxfabric with AutoNode config {}
[2025-03-12 22:12:32,584][ogn_registration][INFO] - Registering Python Node Types from omni.physxfabric at /home/jordan/anaconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.fabric in omni.physx.fabric
[2025-03-12 22:12:32,584][ogn_registration][INFO] - ========================================================================================================================
[2025-03-12 22:12:32,585][ogn_registration][INFO] - No dependency on omni.graph, therefore no nodes to register in omni.physx.fabric
[2025-03-12 22:12:32,585][ogn_registration][INFO] - ...None found, no registration to do
[2025-03-12 22:12:32,585][ogn_registration][INFO] - ...Skipping: No OmniGraph presence in the module omni.physxfabric - No nodes in this module, do not remember it
[2025-03-12 22:12:32,585][ogn_registration][INFO] - Destroying registration record for omni.physx.fabric
[2025-03-12 22:12:32,585][ogn_registration][INFO] - OGN register omni.physx.fabric-106.5.7 took 1105016.000000
[INFO]: Base environment:
    Environment device    : cuda:0
    Environment seed      : 100
    Physics step-size     : 0.005
    Rendering step-size   : 0.02
    Environment step-size : 0.02
2025-03-13 02:12:33 [9,981ms] [Warning] [isaaclab.terrains.terrain_importer] Visual material specified for ground plane but no diffuse color found. Using default color: (0.0, 0.0, 0.0)
[INFO]: Time taken for scene creation : 0.893014 seconds
[INFO]: Scene manager:  <class InteractiveScene>
    Number of environments: 4
    Environment spacing   : 2.5
    Source prim name      : /World/envs/env_0
    Global prim paths     : ['/World/ground']
    Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
```

RandomOakForest, 2025-03-14:
Thanks for posting this. Does the simulation eventually resumes? does it work well when running from scratch? We may have to take a closer look.

WangYCheng23, 2025-03-17:
I encountered the same issue. The simulation was stuck at '[INFO]: Starting the simulation. This may take a few seconds. Please wait...' for quite a while before continuing.

RandomOakForest, 2025-04-15:
Typically, Isaac Sim will take time when launching it for the first time. I will close this for now. If this becomes burdensome or there is new information you can share about this slowdown, please do upon a new bug report. Thanks.

closed 2025-04-15 by RandomOakForest