isaac-sim/IsaacLab #2074: [Question] Contact Sensor force not showing x,y directional force

hojae-io, 2025-03-14:
Hi,

I'm using contact sensor for my humanoid foot.
But when I retrieve the contact force, I only detect z-directional force and x,y-directional force is nearly 0.
But obviously there should be x,y-directional frictional force exerted on the foot.

I don't know if this is just inherent problem of contact sensor where it can only detect normal force?
Or I'm using incorrectly?

For more information,
the collision geometry of my foot is long-thin cylinder.

And my code is `contact_sensor.data.net_forces_w`

And I get the value as below:

tensor([[0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [7.9538e-05, 0.0000e+00, 6.6721e+02],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [6.4743e-05, 0.0000e+00, 5.4310e+02],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00],
        [0.0000e+00, 0.0000e+00, 0.0000e+00]])

Please let me know if there's better way to detect the 'correct' xyz contact force for foot.

Thank you!

RandomOakForest, 2025-03-14:
Following up, this is an Isaac Sim issue, and the forces are not being updated as expected. We have tagged this for the team to fix.

kousheekc, 2025-03-18:
I have a similar problem. I have a contact sensor attached to a RigidObject and I want to not only measure the normal forces caused by an interaction but also the friction forces caused by dragging the rigid object along a surface. Is this the correct approach to use the ContactSensor from IsaacLab?

[image]

Mayankm96, 2025-03-19:
As mentioned in the [docs](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sensors.html#isaaclab.sensors.ContactSensorData.net_forces_w), `net_forces_w` is the net contact forces in world frame which is what you see above.

We are working on exposing the friction data in the contact sensor (@jtigue-bdai for vis.). However, if you want to directly get access to it, you can do so via:

```python
sensor.contact_physx_view.get_friction_data(dt)
```

jtigue-bdai, 2025-03-19:
One thing to note: get_friction_data gets data in a different output format so you will need to do some combining to get it in the same format at net_forces_w.

hojae-io, 2025-03-20:
Still, I've tried get_friction_data, but it only has all 0 values.
Maybe would be nice to add some tutorial how to use `get_friction_data` method

hojae-io, 2025-03-26:
[image]

I would appreciate it if we could access all the contact forces (including normal and friction force as well as point of contact) easily, just like they are being used in Mujoco as in figure