isaac-sim/IsaacLab #2076: Import URDF colliders not matching the robot wheels

ecemsumer, 2025-03-14:
I am trying to convert unitree go2w wheeled legged robot urdf to usd to use in isaaclab. I follow the instructions given in the URDF import tutorial (https://docs.isaacsim.omniverse.nvidia.com/latest/robot_setup/import_urdf.html). The wheel colliders look like the photo below when I do the following steps. 

Select: the eye icon in the upper left corner of the viewport.
Select: Show by type.
Select: Physics.
Select: Colliders.
Check: All.

[image]

I am using Isaacsim 4.5.0 on Windows 11.  
The go2w URDF is taken from the github repo provided by unitree robotics, and the same repo is also given as a reference in this repo for unitree robots.

RandomOakForest, 2025-03-14:
Thank you for posting this. It may be an issue with the importer. We are taking a closer look.

RandomOakForest, 2025-03-17:
Following up on this. It seems you may need to rig your robot before using it in simulation. [This tutorial may help](https://docs.isaacsim.omniverse.nvidia.com/latest/robot_setup/rig_mobile_robot.html#rig-a-mobile-robot). If you need further help, I will move this post to a discussion for the team to follow up.

ecemsumer, 2025-03-18:
I corrected the collision meshes by changing the urdf collision frame orientations. Thanks, I appreciate your interest!

closed 2025-03-18 by ecemsumer