isaac-sim/IsaacLab #2078: [Question] Can animation graph be used in Isaac Lab?

shaoxiang, 2025-03-14:
### Question

I wanted to have some human animations when training, so I implemented it in Isaac Sim through animation graph. 
I saved it as a usd file, brought it to Isaac Lab, and added it through **AssetBaseCfg**, but when the scene was loaded, the animation did not play.

[image]

I would like to know whether Isaac Lab supports animation graph? 
If so, what is the correct way to use it? 
Thank you very much.

RandomOakForest, 2025-03-14:
Thanks for posting this. Great question. We don't support action graphs nor Anim Graph in Isaac Lab, currently.

RandomOakForest, 2025-03-17:
Thanks for your interest in Isaac Lab and for volunteering your efforts towards this. The team has delayed supporting such graphs because it is likely to impact overall training performance, as it may need to be done only on CPU. Is there a use case you think would benefit from this?

Mayankm96, 2025-03-24:
Can you change the device to cpu and disable fabric to see if that resolves the issue?

You can do this by modifying the settings in the `SimulationCfg`.

closed 2025-04-15 by RandomOakForest