isaac-sim/IsaacLab #2079: [Question] Clarification on Curriculum Update Timing

ozhanozen, 2025-03-14:
Hello,

I have observed that curriculum elements in IsaacLab are only computed when at least one environment resets. Is this the intended behavior?

This seems counterintuitive to me because curriculum term updates do not necessarily rely on reset information and could be applied across all environments simultaneously. For example, consider the following function:

https://github.com/isaac-sim/IsaacLab/blob/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9/source/isaaclab/isaaclab/envs/mdp/curriculums.py#L21-L36
I would expect the term weight to be updated as soon as `env.common_step_counter > num_steps`. However, if no environment resets for a while, the update does not occur.

This issue becomes particularly noticeable when using custom curriculum terms that modify elements progressively based on `env.common_step_counter`. The effective task difficulty differs across learning libraries:
	•	rsl_rl has `init_at_random_ep_len=True`, effectively triggering curriculum updates at every environment step.
	•	skrl does not have such randomization, leading to less frequent updates (depending on decimation and episode length).

## Suggested Fix (if unintended behavior)

If this behavior is unintended, I suggest modifying the curriculum update logic by moving:
https://github.com/isaac-sim/IsaacLab/blob/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9/source/isaaclab/isaaclab/envs/manager_based_rl_env.py#L354
out of the `_reset_idx()` and the following `if` block:
https://github.com/isaac-sim/IsaacLab/blob/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9/source/isaaclab/isaaclab/envs/manager_based_rl_env.py#L215-L216
This way, curriculum updates will happen at every step by default. Reset-dependent changes can still be handled within specific term functions if necessary. I haven't tested this myself, though.

Looking forward to your insights!

RandomOakForest, 2025-03-14:
Thank you for posting this. I'm reviewing with the team which is the intended behavior.

ozhanozen, 2025-03-19:
> It is intended as the curriculum implemented is a game-based curriculum where at the end of the episode, the performance of the agent is evaluated and its difficulty is adjusted accordingly.
> 
> You are right that if you want this to be triggered based on the simulation counter, then it does not happen properly. The proper solution might be to introduce different curriculum modes similar to event modes that get called based on the desired type of triggering.
> 
> What do you think?

Yes, having different curriculum modes is a great idea! 

In fact, beyond episode-based or simulation-step-based triggers, I found the need for two additional types of triggers in my work:

1. Performance-based triggers: These would adjust the curriculum based on the episode's mean/sum of a specific reward term. Implementing this should be feasible, but having some helper functions in the reward manager could make it more convenient and standardized.

2. Algorithm-centric (e.g., ppo) triggers: For example, in an insertion task with low tolerances, the standard deviation of the action distribution can aid initial exploration for reaching but hinder precision during insertion. A curriculum could be designed to introduce the insertion phase only when the action distribution’s std falls below a certain threshold. However, integrating such an agent-centric curriculum isn’t straightforward, as the agent and environment are currently quite decoupled.

What are your thoughts on these?

RandomOakForest, 2025-04-18:
Thanks for following up. I will move this to our _Discussions_ for the team to continue if they have any additional feedback. Thanks for your interest in Isaac Lab. @Mayankm96 for vis. Thanks.