isaac-sim/IsaacLab #2081: [Bug Report] Bug title

zhimou, 2025-03-14:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I have upgrade my graphic card to 5070Ti, and install isaaclab as instruction. It can run the headless training example well. However, When I try to remove the "--headless", it crashed with an "invalid value for --gpu-architecture (-arch)" error.

### Steps to reproduce

My step to install isaac lab:
./isaaclab.sh -i
pip install --upgrade --pre torch torchvision torchaudio --index-url https://download.pytorch.org/whl/nightly/cu128 
# I added from the instruction torchvision and torchaudio otherwise I get other error from torchvision

./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0

```
[INFO]: Base environment:
	Environment device    : cuda:0
	Environment seed      : 42
	Physics step-size     : 0.005
	Rendering step-size   : 0.02
	Environment step-size : 0.02
[INFO] Generating terrains based on curriculum took : 1.450720 seconds
[INFO]: Time taken for scene creation : 12.186772 seconds
[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 4096
	Environment spacing   : 2.5
	Source prim name      : /World/envs/env_0
	Global prim paths     : ['/World/ground']
	Replicate physics     : True
[INFO] Event Manager:  <EventManager> contains 3 active terms.
+--------------------------------------+
| Active Event Terms in Mode: 'startup' |
+----------+---------------------------+
|  Index   | Name                      |
+----------+---------------------------+
|    0     | physics_material          |
|    1     | add_base_mass             |
+----------+---------------------------+
+---------------------------------------+
|  Active Event Terms in Mode: 'reset'  |
+--------+------------------------------+
| Index  | Name                         |
+--------+------------------------------+
|   0    | base_external_force_torque   |
|   1    | reset_base                   |
|   2    | reset_robot_joints           |
+--------+------------------------------+
+----------------------------------------------+
|    Active Event Terms in Mode: 'interval'    |
+-------+------------+-------------------------+
| Index | Name       | Interval time range (s) |
+-------+------------+-------------------------+
|   0   | push_robot |       (10.0, 15.0)      |
+-------+------------+-------------------------+

[INFO]: Starting the simulation. This may take a few seconds. Please wait...
2025-03-14 14:01:55 [56,575ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/envs/env_0/Robot/LF_FOOT/visuals.proto_mesh_0_id0 that has not been populated
2025-03-14 14:01:55 [56,693ms] [Warning] [omni.hydra] Mesh '/World/envs/env_0/Robot/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st': buffer size 702 doesn't match expected size 12828 in faceVarying primvars
[INFO]: Time taken for simulation start : 40.415052 seconds
[INFO] Command Manager:  <CommandManager> contains 1 active terms.
+------------------------------------------------+
|              Active Command Terms              |
+-------+---------------+------------------------+
| Index | Name          |          Type          |
+-------+---------------+------------------------+
|   0   | base_velocity | UniformVelocityCommand |
+-------+---------------+------------------------+

[INFO] Recorder Manager:  <RecorderManager> contains 0 active terms.
[... 97 lines cut ...]
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-03-14 14:02:04 [65,389ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-03-14 14:02:04 [65,394ms] [Warning] [omni.graph.core.plugin] Could not find category 'animation' for removal
2025-03-14 14:02:04 [65,908ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-03-14 14:02:05 [66,349ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-03-14 14:02:05 [66,568ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

-->

### System Info

Describe the characteristic of your environment:

- Commit: [91f53e2]
- Isaac Sim Version: [4.5.0-rc.36+release.19112.f59b3005.gl]
- OS: [Ubuntu 20.04]
- GPU: [RTX 5070 Ti]
- CUDA: [12.8]
- GPU Driver: [570.124.04 open]

sincerem00n, 2025-03-14:
hi zhimou,

i got the same problem as you after 7 epoch of training. i tried training with Anymal C which used to work well (1 month ago) but still got the same problem with same script i used `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --max_iterations=1000 --num_env=32 --headless --video`

have you found a solution for it yet?

zhimou, 2025-03-15:
> hi zhimou,
>
> i got the same problem as you after 7 epoch of training. i tried training with Anymal C which used to work well (1 month ago) but still got the same problem with same script i used [...]

I am able to run your command in my desktop using open nvidia driver 570 and cuda 12.8. You can try to purge your nvidia driver then reinstall it, then try my installation instruction above. But I can't run anything without "--headless" thougth

sincerem00n, 2025-03-15:
> > hi zhimou,
> > i got the same problem as you after 7 epoch of training. i tried training with Anymal C which used to work well (1 month ago) but still got the same problem with same script i used `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --max_iterations=1000 --num_env=32 --headless --video`
> > have you found a solution for it yet?
> 
> I am able to run your command in my desktop using open nvidia driver 570 and cuda 12.8. You can try to purge your nvidia driver then reinstall it, then try my installation instruction above. But isaac sim will crash without "--headless" thougth

I try training on my window 11 laptop with RTX4070 using nvidia driver 572.70 and cuda 11.8. It works well without any crashing. I don't know why attempting to train the model on linux server didn't worked...

ChengJinyu6, 2025-03-24:
#

> > > > hi zhimou,  嗨，志谋，
> > > > i got the same problem as you after 7 epoch of training. i tried training with Anymal C which used to work well (1 month ago) but still got the same problem with same script i used `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --max_iterations=1000 --num_env=32 --headless --video`经过 7 个时期的训练后，我遇到了和你一样的问题。我尝试使用 Anymal C 进行训练，它曾经运行良好（1 个月前），但仍然在使用我使用的 `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --max_iterations=1000 --num_env=32 --headless --video` 相同脚本时遇到同样的问题
> > > > have you found a solution for it yet?您找到解决方案了吗？
> > > 
> > > 
> > > I am able to run your command in my desktop using open nvidia driver 570 and cuda 12.8. You can try to purge your nvidia driver then reinstall it, then try my installation instruction above. But isaac sim will crash without "--headless" thougth我能够使用打开的 nvidia 驱动程序 570 和 cuda 12.8 在我的桌面上运行您的命令。您可以尝试清除您的 nvidia 驱动程序，然后重新安装它，然后尝试我上面的安装说明。但是 isaac sim 在没有 “--headless” 的情况下会崩溃
> > 
> > 
> > I try training on my window 11 laptop with RTX4070 using nvidia driver 572.70 and cuda 11.8. It works well without any crashing. I don't know why attempting to train the model on linux server didn't worked...我尝试使用 NVIDIA 驱动程序 572.70 和 cuda 11.8 在我的 Windows 11 笔记本电脑上使用 RTX4070 进行训练。它运行良好，没有任何崩溃。我不知道为什么尝试在 linux 服务器上训练模型不起作用......
> 
> It has something to do with 570 driver in linux and cuda 12.8. My 40 series card works fine in my other machine它与 linux 和 cuda 570 中的 cuda 12.8 驱动程序有关。我的 40 系列卡在另一台机器上工作正常

Hi, do you fix that issue? i have the same issue now, when i run training , the window crashed and give a nvrtc compilation failure.

zhimou, 2025-03-26:
> Hi,
> 
> I was able to run IsaacLab without the `--headless` flag by upgrading CUDA 12.8 and uninstalling outdated CUDA 11.8-related Python packages.
> 
> To check if these packages are present, activate the `env_isaaclab` conda environment and run:
> 
> pip list | grep -i cuda
> If you find CUDA 11 packages, you can remove them using:
> 
> pip uninstall nvidia-cuda-cupti-cu11  
> pip uninstall nvidia-cuda-nvrtc-cu11  
> pip uninstall nvidia-cuda-runtime-cu11
> After doing this, I was able to run IsaacLab successfully. However, I am unsure of the impact on compatibility with other IsaacSim extensions. Proceed with caution.

Man, you are legend. It solved my issue! For some reason this cuda 11 package is automatically install in my conda new isaac environment. Now they are displaying fine.

zhimou, 2025-03-26:
> Hi, do you fix that issue? i have the same issue now, when i run training , the window crashed and give a nvrtc compilation failure.

riccardobussola solved my issue in below.

closed 2025-03-26 by zhimou

81578823, 2025-04-18:
> Hi,
>
> I was able to run IsaacLab without the `--headless` flag by upgrading CUDA 12.8 and uninstalling outdated CUDA 11.8-related Python packages.
>
> To check if these packages are present, activate [...]

you are a hero, man. Thanks

81578823, 2025-04-18:
> Hi,
>
> I was able to run IsaacLab without the `--headless` flag by upgrading CUDA 12.8 and uninstalling outdated CUDA 11.8-related Python packages.
>
> To check if these packages are present, activate [...]

You are a hero, man! Thank you!