isaac-sim/IsaacLab #2084: Isaacsim pip install: No module named 'isaacsim.simulation_app'

toenails6, 2025-03-14:
### Describe the bug

After following install instructions with conda in windows, attempting to launch Isaac sim in powershell with command isaacsim yields: 
No module named 'isaacsim.simulation_app'
Specifically: 

[image]

### Steps to reproduce

Directly followed newest Isaac sim pip install instructions with conda in Windows, CUDA 12. 

```
Traceback (most recent call last):
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\site-packages\isaacsim\__init__.py", line 90, in expose_api
    from isaacsim.simulation_app import AppFramework, SimulationApp
ModuleNotFoundError: No module named 'isaacsim.simulation_app'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\renda\anaconda3\envs\isaaclab\Scripts\isaacsim.exe\__main__.py", line 4, in <module>
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\site-packages\isaacsim\__init__.py", line 170, in <module>
    AppFramework, SimulationApp = expose_api()
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\site-packages\isaacsim\__init__.py", line 96, in expose_api
    path = glob.glob(
IndexError: list index out of range
```

### System Info

Describe the characteristic of your environment:

- Commit: b5fa0eb03
- Isaac Sim Version: newest official 4.5.0
- OS: Windows
- GPU: RTX 3060
- CUDA: 12.1
- GPU Driver: 572.60

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Able to launch isaacsim from powershell
- [ ] Able to use Isaac Lab as normal

Toni-SM, 2025-03-14:
Hi @toenails6 

Could you please share the exactly commands you executed to install Isaac Sim / Lab?
Also, what is the `pip list` output?

toenails6, 2025-03-14:
`conda create -n isaaclab python=3.10`
`pip install torch==2.5.1 --index-url https://download.pytorch.org/whl/cu121`
`python -m pip install --upgrade pip`
`pip install 'isaacsim[all,extscache]==4.5.0' --extra-index-url https://pypi.nvidia.com`

`pip list` yields the following: 
```
boto3                      1.37.13
botocore                   1.37.13
cloudpickle                3.1.1
contourpy                  1.3.1
cycler                     0.12.1
Farama-Notifications       0.0.4
filelock                   3.13.1
fonttools                  4.56.0
fsspec                     2024.6.1
gymnasium                  1.1.1
isaacsim                   4.5.0.0
isaacsim-app               4.5.0.0
isaacsim-asset             4.5.0.0
isaacsim-benchmark         4.5.0.0
isaacsim-code-editor       4.5.0.0
isaacsim-core              4.5.0.0
isaacsim-cortex            4.5.0.0
isaacsim-example           4.5.0.0
isaacsim-extscache-kit     4.5.0.0
isaacsim-extscache-kit-sdk 4.5.0.0
isaacsim-extscache-physics 4.5.0.0
isaacsim-gui               4.5.0.0
isaacsim-kernel            4.5.0.0
isaacsim-replicator        4.5.0.0
isaacsim-rl                4.5.0.0
isaacsim-robot             4.5.0.0
isaacsim-robot-motion      4.5.0.0
isaacsim-robot-setup       4.5.0.0
isaacsim-ros1              4.5.0.0
isaacsim-ros2              4.5.0.0
isaacsim-sensor            4.5.0.0
isaacsim-storage           4.5.0.0
isaacsim-template          4.5.0.0
isaacsim-test              4.5.0.0
isaacsim-utils             4.5.0.0
Jinja2                     3.1.4
jmespath                   1.0.1
kiwisolver                 1.4.8
MarkupSafe                 2.1.5
matplotlib                 3.10.1
mpmath                     1.3.0
networkx                   3.3
numpy                      1.26.4
omniverse-kit              106.5.0.162521
opencv-python              4.11.0.86
osqp                       0.6.7.post3
packaging                  24.2
pillow                     11.1.0
pip                        25.0.1
psutil                     7.0.0
pyparsing                  3.2.1
pyperclip                  1.9.0
python-dateutil            2.9.0.post0
PyYAML                     6.0.2
qdldl                      0.1.7.post5
s3transfer                 0.11.4
scipy                      1.15.2
setuptools                 75.8.0
six                        1.17.0
sympy                      1.13.1
torch                      2.5.1+cu121
typing_extensions          4.12.2
urllib3                    2.3.0
wheel                      0.45.1
```

toenails6, 2025-03-15:
Oh boy, for some reason it works on my Alienware M18 R2, but not on my RTX 3060 desktop. 
I'm still trying to figure out why this could possibly be happening.

toenails6, 2025-03-15:
The issue persists, I am going to resort to SSH remote editing for now. My desktop has more monitors so writing code is easier, but I have to run IsaacLab on my Alienware by this point. 
I would still be very interested in getting this to work indigenously on my desktop.

Toni-SM, 2025-03-15:
Hi @toenails6 

Could you please try deactivating all the conda environments and installing Isaac Sim using Python virtual environment (venv)?

toenails6, 2025-03-15:
I don't have python3.10 installed in my windows environment, and I just checked, seems a hassle to try to do so.

Toni-SM, 2025-03-15:
Hi @toenails6 

You can get Python3.10 from Window App Store or download and install it from https://www.python.org/downloads/windows/

toenails6, 2025-03-17:
Nah, python virtual environment does not work either. 
Yields similar error outputs. 

[image]

toenails6, 2025-03-17:
By the way, I did already enable long paths on both my computers, so I do not think that would be the issue.

toenails6, 2025-03-17:
I messed around a little, and all I can say is: 
In the `expose_api()` function, both my computers go to getting the simulation_app from path. 
From here, everything works fine until the `glob.glob()` function. 
Both my alienware and desktop can get the correct `isaacsim_path` values. 
But then, my desktop somehow executes the `glob.glob()` function erroneously, but my alienware does fine. 
Since the `isaacsim_path` variable is correct, there's likely something off with my desktop's `os` package or the `glob.glob()` function. 
This is all I can do on my own.

toenails6, 2025-03-17:
I confirmed it with some rudimentary print debugging, the outputs from `os.environ.get` is broken. 
The following is from my desktop: 

[image]

toenails6, 2025-03-17:
This then traces up to the `bootstrap_kernel()` function: 

[image]

Let me keep looking.

toenails6, 2025-03-17:
Weird, the `os.environ.get()` should yield the value of `isaacsim_path` even `bootstrap_kernel()` failed to set `ISAAC_PATH` before hand though. 
This is exceptionally out of hand.

toenails6, 2025-03-17:
I see, `os.environ.get()' did retrieve some value, but that value was erroneous in the first place. 
Which still should trace back up to `bootstrap_kernel()`.

toenails6, 2025-03-17:
Damn, I locked it down, there's a ghost system environment path variable that I must have left around back during my binary install days:
[image]

toenails6, 2025-03-17:
The ghost environment variable is left from a previous binary installation attempt/process, I must have appended it to some anaconda powershell startup script somewhere. These startup scripts leaves the environment variable there even after conda deactivation so python virtual environment didn't fix anything. 
Reinstalled anaconda and removed this ghost environment variable. 
Works on my desktop now. 

[image]

[image]

closed 2025-03-17 by toenails6