isaac-sim/IsaacLab #2085: [Bug Report] GPU memory keeps increasing when using object with complex collision mesh

supersglzc, 2025-03-14:
Same as the title. When I use an object with kinda of complicated collision mesh, the GPU memory keeps exploding until reach the limit. Then the program is not killed but continues running with very low GPU utilization and becomes extremely slow. Is there any similar bug seen before?

RandomOakForest, 2025-03-18:
Thanks for posting this. Could you please fill in the template under _Bug Report_ and share code we can use to reproduce this issue? Thanks!

RandomOakForest, 2025-03-18:
Following up on this, it seems this may be expected depending on the mesh. [Isaac Sim has these hardware requirements](https://docs.isaacsim.omniverse.nvidia.com/4.5.0/installation/requirements.html). It may be helpful to post about this in the [Isaac Sim forums](https://forums.developer.nvidia.com/c/omniverse/simulation/69).

closed 2025-04-07 by RandomOakForest