isaac-sim/IsaacLab #2087: [Question] Why is the quality of isaacsim rendering and video output so poor when using rsl_rl's play.py and --video?

xliu0105, 2025-03-15:
### Question

I found that when using --video in play and want to output a demo video, the rendering quality of the isaacsim window is very poor, and the rendering quality of the video output is also very poor. Is there any way to optimize it?

kellyguo11, 2025-03-17:
Thanks for reporting this. Render settings can be modified through `RenderCfg`. Could you share more details on which task are you trying to capture?

KyleM73, 2025-03-17:
I've noticed this as well; with fast moving objects there is often ghosting where you can see the trail of last positions for a few frames, especially for debug_viz markers like contact sensors or velocity arrows in the quadruped locomotion environments. I haven't tried playing with the render config settings too much, but changing the anti-aliasing to TAA helped I believe. I can try testing more later this week/next week when I have more bandwidth unless someone can get to it first.

xliu0105, 2025-03-18:
> Thanks for reporting this. Render settings can be modified through `RenderCfg`. Could you share more details on which task are you trying to capture?

I think it has nothing to do with the `RenderCfg` parameters. Because if I don't use `--video`, the rendering quality of isaacsim is normal. But if I set `--video`, the rendering quality becomes very bad.

xliu0105, 2025-03-18:
[image]
[image]

This is the rendering when I use `--video` and not use `--video`. It can be seen that the rendering quality is obviously worse when using `--video`.

xliu0105, 2025-03-25:
Thanks for your reply, I merged the latest Isaaclab version and found this issue was fixed. Thank you very much for your help.

closed 2025-04-15 by RandomOakForest