isaac-sim/IsaacLab #2089: [Bug Report] [Error] [omni.physx.tensors.plugin] Incompatible device of velocity tensor in function getVelocities: expected device 0, received device -1

shiven2001, 2025-03-16:
### Describe the bug
I am trying to use the tutorial nova_carter_ros.usd from the isaacsim in issaclab. I created the environment and set my simulation cfg as:
`
    sim_cfg = sim_utils.SimulationCfg(dt=0.01, device="cuda")
    sim = SimulationContext(sim_cfg)
`
But when I do this, I get this error:

[Error] [omni.physx.tensors.plugin] Incompatible device of velocity tensor in function getVelocities: expected device 0, received device -1

The nova_carter_ros.usd has many sensors on it including 3d lidar, cameras, and imu. If I use the device as cpu:

`
    sim_cfg = sim_utils.SimulationCfg(dt=0.01, device="cpu")
    sim = SimulationContext(sim_cfg)
`
I do not get the error. I am guessing the issue is that nova_carter_ros.usd sensors are using cpu tensor instead gpu based. How to fix this issue? I want to run simulations in parallel for RL. 

Someone opened this issue in 2024 but it was closed saying that imu was the issue and a MR will be done on it but the issue is still here.

### System Info

Issac Sim v4.5.0

Issac Lab v2.0.0

ROS2 Humble

Ubuntu Version 22.04 Jazzy

NVIDIA NVRM Version Kernel Module 570.86.10

GCC version 11.4.0 (Ubuntu 11.4.0 ~22.04)

Cuda Compilation Tools, Release 12.8, V12.8.61

### Checklist

- [X ] I have checked that there is no similar issue in the repo (**required**)
- [X ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

shiven2001, 2025-03-18:
Yes, that's the one!

tberg1234, 2025-03-26:
I am having the same issue, but when I switch to CPU I get warnings of "OrthogonalizeBasis did not converge, matrix may not be orthonormal." and errors "[omni.physx.plugin] PhysX error: Illegal BroadPhaseUpdateData 
, FILE /builds/omniverse/physics/physx/source/lowlevelaabb/src/BpBroadPhaseABP.cpp, LINE 4023", and the robots disappear upon spawning.

RandomOakForest, 2025-04-18:
Thanks for following up. The team is reviewing this issue and we're aiming to resolve it as time permits.

amariaholson, 2025-06-18:
Same error for me... i make a xform prim, add some simple child primitives to it, set the xfrom collision on...

Import into isaaclab and get these errors:

2025-06-18 05:42:32 [39,035ms] [Error] [omni.physx.tensors.plugin] Incompatible device of velocity tensor in function getVelocities: expected device 0, received device -1

If I replace the import with a simple ball shape primitive instead, the error goes away... it looks like that isaac SIM is  building it on CPU. IN isaaclab somehow we need to move it to GPU? HOW?

kellyguo11, 2025-06-25:
We are working together with the Isaac Sim team on this. It's a bit more complex because the current ROS bridge does not support GPU pipelines.

Dafodilrat, 2025-08-20:
I am getting the same issue however I am not using ROS. This is happening while trying to run an external isaac lab project on a DirectRLEnv

matthewtrepte, 2026-10-19:
This error comes from Isaac Sim's physx tensors plugin, and the original report was on Isaac Sim 4.5 / Isaac Lab 2.0. Closing as stale. If it still happens on Isaac Lab 3 with a current Isaac Sim, please reopen

closed 2026-10-19 by matthewtrepte