isaac-sim/IsaacLab #2092: [Question] python IsaacLab/source/isaaclab/test/deps/isaacsim/check_rep_texture_randomizer.py throws an error.

idigitopia, 2025-03-16:
### Question
    from isaacsim.core.objects import DynamicSphere
ModuleNotFoundError: No module named 'isaacsim.core.objects'

objects not found in isaacsim

RandomOakForest, 2025-03-18:
Thanks for posting this. This is typically an issue with set up, or that you are not launching the omniverse application first. Do you have the following before your import?
```
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app
```

idigitopia, 2025-03-21:
from isaacsim.core.api.materials import PhysicsMaterial, PreviewSurface
from isaacsim.core.api.objects import DynamicSphere

I fixed it by chanigng the imports. I think the imports  were wrong. it has to be isaacsim.core.api.objects now.

closed 2025-04-15 by RandomOakForest