isaac-sim/IsaacLab #2093: [Bug Report] Bug title

ecstayalive, 2025-03-17:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-03-18:
Thank you for posting this. The team will review this.

qiaoyo, 2025-05-29:
Hi, you can fix it by modifying the file "pyproject.toml", add following works for me :
`[tool.pyright]
reportMissingImports = true  
reportUnusedVariable = true  

reportArgumentTypeMismatch = false`

ecstayalive, 2025-06-06:
> Hi, you can fix it by modifying the file "pyproject.toml", add following works for me : `[tool.pyright] reportMissingImports = true reportUnusedVariable = true
> 
> reportArgumentTypeMismatch = false`

I've tried your method, but it only removes a portion of the type error

qiaoyo, 2025-07-10:
> > Hi, you can fix it by modifying the file "pyproject.toml", add following works for me : `[tool.pyright] reportMissingImports = true reportUnusedVariable = true
> > 
> > reportArgumentTypeMismatch = false`
> 
> I've tried your method, but it only removes a portion of the type error

that's for sure. this is the way to remove the error hint. you can add more to remove