isaac-sim/IsaacLab #2094: [Bug Report] UnicodeEncodeError: 'charmap' codec can't encode character '\u2630' in position 3014: character maps to <undefined>

toenails6, 2025-03-17:
### Describe the bug

Attempt to generate external project leads to the mentioned error. 

### Steps to reproduce

Any common step to generate an external project. 

```
Validating specification...
Generating external project...
  |-- Copying repo files...
Traceback (most recent call last):
  File "C:\Users\renda\IsaacLab\tools\template\cli.py", line 261, in <module>
    main()
  File "C:\Users\renda\IsaacLab\tools\template\cli.py", line 257, in main
    generate(specification)
  File "C:\Users\renda\IsaacLab\tools\template\generator.py", line 283, in generate
    _external(specification)
  File "C:\Users\renda\IsaacLab\tools\template\generator.py", line 171, in _external
    _write_file(os.path.join(project_dir, "README.md"), content=template.render(**specification))
  File "C:\Users\renda\IsaacLab\tools\template\generator.py", line 64, in _write_file
    file.write(content)
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u2630' in position 3014: character maps to <undefined>
```

### System Info

Describe the characteristic of your environment:

- Commit: 91f53e2fe
- Isaac Sim Version: newest official 4.5.0
- OS: Windows
- GPU: RTX 4090
- CUDA: 12.1
- GPU Driver: 572.60

### Additional context

System has Chinese language pack, but the primary and display language is set to English. 
Though Microsoft seems to be having some bugs with the backend language priorities lately. 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Generates external project as intended.

RandomOakForest, 2025-03-18:
Thank you for posting this. Could you share the command line argument you launched that gave you this error? please elaborate on how are you trying to run Isaac Lab. Thanks.

toenails6, 2025-03-18:
Attached is the full prompt and process: 

[image]

toenails6, 2025-03-18:
I also tried changing the project name and adding '\' to the end of project path but nothing works.

Toni-SM, 2025-03-18:
Hi @toenails6 

While we investigate it, could you please comment in `C:\Users\renda\IsaacLab\tools\template\generator.py` the line 171 (it just generate the README) and try again and see if you get other error?

toenails6, 2025-03-18:
Everything else generated (except for the README obviously). 
Yeah, the read me generator is the only thing bugged as of yet.

toenails6, 2025-03-18:
Ah yea I briefly remember that icon from googling the unicode. 
But there shouldn't be anything I should do on my end right now, right? 
Also, I briefly remember there was some solution for some other thing regarding using a unicode output encoding that supports a broader range of symbols.

closed 2025-03-19 by kellyguo11