isaac-sim/IsaacLab #2095: Unit of measure mismatch for Intel Realsense D455 camera

torydebra, 2025-03-17:
Hi all,

I was checking the [doc](https://docs.isaacsim.omniverse.nvidia.com/latest/assets/usd_assets_sensors.html#intel) for the D455 camera, params are set like: 
| name    | Camera_Pseudo_Depth |
| -------- | ------- |
| focalLength  | 1.9299999475479126    |

values that are also visible in the simulation while importing the camera from the assets:
[image]

According to the realsense [datasheet](https://www.intelrealsense.com/wp-content/uploads/2024/10/Intel-RealSense-D400-Series-Datasheet-October-2024.pdf), 1.93 is in **mm**

But in the [pinhole camera model](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.spawners.html#isaaclab.sim.spawners.sensors.PinholeCameraCfg) (the way the camera is simulated) doc it is stated that values should be entered in centimeters:
```
focal_length: [float]
Perspective focal length (in cm). Defaults to 24.0cm.
```

Furthermore, the same doc state this:
```
Focal length as well as the aperture sizes and offsets are set as a tenth of the world unit. In our case, the world unit is Meter s.t. all of these values are set in cm.
```
and if I am not wrong a *tenth* of meter is dm (0.1) not cm (0.01).

So overall, who is right? Values should be entered in **dm**, **cm**, or **mm**?

torydebra, 2025-03-18:
Hi, thanks for the quick response

So if everything is in cm, is the camera model wrong? Since the actual numbers (like **focalLength 1.9**) refers to **mm**, not cm

Also, is the doc of the replicator library that is referenced in the issue you linked wrong as well? It states:
```
0.1 * world units
```

pascal-roth, 2025-04-03:
So the IsaacSim documentation states "cm (or 1/tenth of the world unit)" ([Source](https://docs.omniverse.nvidia.com/isaacsim/latest/features/sensors_simulation/isaac_sim_sensors_camera.html#calibrated-camera-sensors)), while the original USD documentation is more clear and states 1/tenth of a scene unit ([Source](https://openusd.org/dev/api/class_usd_geom_camera.html#a15709b0274673a367bcac5d4978500aa)). 

I checked for different focal lengths and could reproduce the reported issue when it is set as 1/tenth of the world unit, e.g., 1.93mm --> 0.193 in the code. The issue seems to be coming from IsaacSim site, will forward the issue to the IsaacSim team. 

Meanwhile, I recommend setting the focal length to mm and, ideally, using the intrinsic matrix.

closed 2025-04-15 by RandomOakForest