isaac-sim/IsaacLab #2097: [Question] Contact-rich interactions unstable

shafeef901, 2025-03-17:
### Question

While performing contact-rich interactions using an Allegro hand and a simple cuboid with small mass and high friction, the contact interactions seem too unstable. Training such a policy also results in the `action_l2` and `action_rate_l2` penalties to blow up. Is there anything that can be done to make the simulation more stable when dealing with such contact-rich interactions? I tried changing a bit the values of `solver_position_iteration_count`, `solver_velocity_iteration_count`, and `max_depenetration_velocity`, but my efforts were in vain. 

Also, I couldn't find any documentation on this. It'd be great to have some details about it in the docs.

https://github.com/user-attachments/assets/fab19f63-436d-425f-b5c9-907ff4f2c5c6

Action Penalties Plots:
[image]
Mean Reward Plots:
[image]

Configuration of the object:
```python
object: RigidObjectCfg = RigidObjectCfg(
        prim_path="{ENV_REGEX_NS}/object",
        spawn=sim_utils.CuboidCfg(
            size=(0.05, 0.12, 0.05),
            activate_contact_sensors=True,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                kinematic_enabled=False,
                disable_gravity=False,
                enable_gyroscopic_forces=True,
                solver_position_iteration_count=8,
                solver_velocity_iteration_count=0,
                sleep_threshold=0.005,
                stabilization_threshold=0.0025,
                max_depenetration_velocity=0.5,
            ),
            mass_props=sim_utils.MassPropertiesCfg(density=70.0),
            physics_material=sim_utils.RigidBodyMaterialCfg(
                static_friction=100.0, dynamic_friction=100.0, friction_combine_mode="max"
            ),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(1.0, 1.0, 0.0), metallic=0.2),
            collision_props=sim_utils.CollisionPropertiesCfg(
                collision_enabled=True,
            ),
        ),
        collision_group=-1,
        init_state=RigidObjectCfg.InitialStateCfg(pos=(-0.09, 0.02, 0.02), rot=(1.0, 0.0, 0.0, 0.0)),
    )
```
Configuration of the allegro hand:
```python
ALLEGRO_HAND_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{CURRENT_WORKING_DIR}/assets/robots/allegro_hand_description/usd/allegro_left.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=True,
            retain_accelerations=False,
            enable_gyroscopic_forces=True,
            linear_damping=0.0,
            angular_damping=0.0,
            max_linear_velocity=1000.0,
            max_angular_velocity=3666.0,
            max_depenetration_velocity=1.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True,
            solver_position_iteration_count=8,
            solver_velocity_iteration_count=0,
            sleep_threshold=0.005,
            stabilization_threshold=0.0025,
        ),
    ),
    actuators={
        "fingers": ImplicitActuatorCfg(
            joint_names_expr=[".*"],
            effort_limit=0.5,
            velocity_limit=100.0,
            stiffness=3.0,
            damping=0.1,
            friction=0.01,
        ),
    },
    soft_joint_pos_limit_factor=1.0,
)
```

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

Mayankm96, 2025-03-17:
Very likely, you might be spawning your objects in penetration which is causing the simulation to go unstable. Also you should reduce the velocity limit for the allegro hand actuator to around 10.0 for a stabler simulation.

shafeef901, 2025-03-18:
Hi @Mayankm96,
Thank you for the quick response. I am pretty sure I am not spawning in penetration, although it is quite close. I've also tried lowering the velocity limit to 1.0 but the same problem persists :/

shafeef901, 2025-03-18:
@noseworm has some experience dealing with contact-rich interactions in IsaacLab (#1520). Maybe he has some insights?

closed 2025-03-26 by shafeef901