isaac-sim/IsaacLab #2101: [Bug Report] Failed to set DOF actuation forces in backend

YT-wk, 2025-03-18:
### Describe the bug

When resetting environments in IsaacLab with CUDA acceleration, a device-side assertion failure occurs due to index out-of-bounds access during DOF actuation force submission. This manifests as CUDA kernel assertion errors (`"index out of bounds"`) followed by failure to set DOF forces in the backend.

### Steps to reproduce

1. **Code Setup**: Use the provided `_reset_idx` method to reset environments with CUDA tensors.
2. **Reset Logic**:
   • Reset robot root/joint states via `write_root_pose_to_sim`, `write_root_velocity_to_sim`, and `write_joint_state_to_sim`.
   • Use `env_ids` to index into default joint/root states.
3. **Trigger Error**:
   • During training, after environment resets, the error occurs when submitting DOF actuation forces via `set_dof_actuation_forces`.

**Minimal Code Snippet**:
```python
    def _reset_idx(self, env_ids: torch.Tensor | None):
        if env_ids is None or len(env_ids) == self.num_envs:
            env_ids = self._robot._ALL_INDICES
        self._robot.reset(env_ids)
        
        super()._reset_idx(env_ids)
        if len(env_ids) == self.num_envs:
            # Spread out the resets to avoid spikes in training when many environments reset at a similar time
            self.episode_length_buf[:] = torch.randint_like(self.episode_length_buf, high=int(self.max_episode_length))
            # reset goal position
            self.goal_positions[env_ids] = self._sample_goals(env_ids)
            

        print("envs need to reset are ",env_ids," 0311************************")

        joint_pos = self._robot.data.default_joint_pos[env_ids]
        joint_vel = self._robot.data.default_joint_vel[env_ids]
        default_root_state = self._robot.data.default_root_state[env_ids]
        default_root_state[:, :3] += self._terrain.env_origins[env_ids]
        print("(joint_pos,num_joints,default_root_state) is",joint_pos,self._robot.num_joints, default_root_state," 0317*********************")
        self._robot.write_root_pose_to_sim(default_root_state[:, :7], env_ids)
        self._robot.write_root_velocity_to_sim(default_root_state[:, 7:], env_ids)
        self._robot.write_joint_state_to_sim(joint_pos, joint_vel, None, env_ids)
        
        # reset goal position
        self.goal_positions[env_ids] = self._sample_goals(env_ids)
        print("reset done. 0315***************")
```

### System Info

• Commit: [IsaacLab commit hash]
• Isaac Sim Version: [4.5.0]
• Isaac Lab Version: [2.0.0]
• OS: [Ubuntu 20.04]
• GPU: [RTX 4080]
• CUDA: [12.8.61]
• GPU Driver: [535.183.01]

### Error Logs

```
envs need to reset are  tensor([10], device='cuda:0')  0311************************
(joint_pos,num_joints,default_root_state) is tensor([[0., 0., 0., 0.]], device='cuda:0') 4 tensor([[-10.,  10.,   0.,   1.,   0.,   0.,   0.,   0.,   0.,   0.,   0.,   0.,
           0.]], device='cuda:0')  0317*********************
reset done. 0315***************
../aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [0,0,0], thread: [0,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [0,0,0], thread: [1,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
../aten/src/ATen/native/cuda/IndexKernel.cu:93: operator(): block: [0,0,0], thread: [2,0,0] Assertion `-sizes[i] <= index && index < sizes[i] && "index out of bounds"` failed.
2025-03-18 01:52:35 [22,866ms] [Error] [omni.physx.tensors.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuArticulationView.cpp: 639
2025-03-18 01:52:35 [22,866ms] [Error] [omni.physx.tensors.plugin] CUDA error: device-side assert triggered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/CudaKernels.cu: 634
2025-03-18 01:52:35 [22,866ms] [Error] [omni.physx.tensors.plugin] Failed to submit DOF actuation force attribute
  0%|                                                                                                                                                             | 17/5000000 [00:05<466:08:47,  2.98it/s]
Traceback (most recent call last):
  File "/media/robot/ef64217c-7820-452d-931f-2253a903882d/robot/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/test250302_navtask/agents/torch_nav_td3.py", line 156, in <module>
    trainer.train()
  File "/home/robot/anaconda3/envs/isaaclab_2_0/lib/python3.10/site-packages/skrl/trainers/torch/sequential.py", line 86, in train
    self.single_agent_train()
  File "/home/robot/anaconda3/envs/isaaclab_2_0/lib/python3.10/site-packages/skrl/trainers/torch/base.py", line 196, in single_agent_train
    next_states, rewards, terminated, truncated, infos = self.env.step(actions)
  File "/home/robot/anaconda3/envs/isaaclab_2_0/lib/python3.10/site-packages/skrl/envs/wrappers/torch/isaaclab_envs.py", line 62, in step
    observations, reward, terminated, truncated, self._info = self._env.step(actions)
  File "/home/robot/anaconda3/envs/isaaclab_2_0/lib/python3.10/site-packages/gymnasium/wrappers/common.py", line 393, in step
    return super().step(action)
  File "/home/robot/anaconda3/envs/isaaclab_2_0/lib/python3.10/site-packages/gymnasium/core.py", line 322, in step
    return self.env.step(action)
  File "/media/robot/ef64217c-7820-452d-931f-2253a903882d/robot/IsaacLab/source/isaaclab/isaaclab/envs/direct_rl_env.py", line 363, in step
    self.scene.write_data_to_sim()
  File "/media/robot/ef64217c-7820-452d-931f-2253a903882d/robot/IsaacLab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 488, in write_data_to_sim
    articulation.write_data_to_sim()
  File "/media/robot/ef64217c-7820-452d-931f-2253a903882d/robot/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 195, in write_data_to_sim
    self.root_physx_view.set_dof_actuation_forces(self._joint_effort_target_sim, self._ALL_INDICES)
  File "/home/robot/anaconda3/envs/isaaclab_2_0/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physics.tensors/omni/physics/tensors/impl/api.py", line 1528, in set_dof_actuation_forces
    raise Exception("Failed to set DOF actuation forces in backend")
Exception: Failed to set DOF actuation forces in backend
[skrl:INFO] Closing environment
[skrl:INFO] Environment closed
```

### Additional context
• **Task description**: I tried to use `skrl` to perform reinforcement learning training in IsaacLab. I used Direct Workflow to build my reinforcement learning environment. I wrote the `_reset_idx()` method based on the `Isaac-Franka-Cabinet-Direct-v0` and `Isaac-Velocity-Flat-Anymal-C-Direct-v0` environments. My robot cfg is:
```
TEST0302_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path="/path/to/my/usd",
        
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            rigid_body_enabled=True,  
            max_linear_velocity=1.0,        
            max_angular_velocity=3600.0,       
            max_depenetration_velocity=100.0,       
            enable_gyroscopic_forces=True,        
        ),
        
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False,     
            solver_position_iteration_count=4,     
            solver_velocity_iteration_count=0,      
            sleep_threshold=0.005,     
            stabilization_threshold=0.001,    
        ),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.0), joint_pos={"j1": 0.0, "j2": 0.0, "j3": 0.0, "j4": 0.0}
    ),
    actuators={
        "fourwheel_actuator": ImplicitActuatorCfg(
            joint_names_expr=["j2","j1","j3","j4"],
            effort_limit_sim=100.0,    
            velocity_limit_sim=100.0,  
            stiffness=0.0,         
            damping=10.0,          
        ),
    },
    prim_path = "/World/envs/env._*/Robot"

)
```
In addition, the following method is used in my scene to generate a new rigid body:
```
obstacle_cfg = sim_utils.MeshCuboidCfg(
    size=size,
    rigid_props=sim_utils.RigidBodyPropertiesCfg(
        rigid_body_enabled=True,
        solver_position_iteration_count=4,
        solver_velocity_iteration_count=1
    ),
    visual_material=sim_utils.PreviewSurfaceCfg(
        diffuse_color=color,
        roughness=0.8,
        metallic=0.2
    ),
    collision_props=sim_utils.CollisionPropertiesCfg(
        collision_enabled=True,
        contact_offset=0.1,
        rest_offset=0.01
    )
)

obstacle_cfg.func(
    prim_path=prim_path,
    cfg=obstacle_cfg,
    translation=position
)
```

• **Hypothesis**: The error likely occurs due to invalid tensor indexing in CUDA kernels during state resetting (e.g., mismatched dimensions between `env_ids` and tensor slices).

### Checklist

• [x] Confirmed no duplicate issues in the repo.
• [x] Verified error occurs in IsaacLab code, not base Isaac Sim.

RandomOakForest, 2025-03-18:
Thank you for posting this. The team will review.

Mayankm96, 2025-03-19:
Can you run your simulation on CPU and see where the error is getting triggered?

You can pass the `--device cpu` to your script.

YT-wk, 2025-03-20:
> Can you run your simulation on CPU and see where the error is getting triggered?
> 
> You can pass the `--device cpu` to your script.

@Mayankm96 Thanks for your reply. I found the problem when I selected `cpu` as the device. In fact, the error was caused by the wrong index somewhere in the `_sample_goals` function.

closed 2025-03-20 by YT-wk