isaac-sim/IsaacLab #2102: How to isolate cameras in parallel environments？

Atticlmr, 2025-03-18:
I need to use camera data for training in Isaac Lab, such as in drone navigation. I have added a camera to each drone in parallel environments, but I want to ensure that drones in one environment (e.g., env1) cannot see drones in another environment (e.g., env2). Specifically, the camera images from a drone in env1 should not capture the drones in env2. How can I achieve this?

Atticlmr, 2025-03-18:
Additionally, I need the drones to see the terrian in their environment to enable obstacle avoidance.

StrainFlow, 2025-03-19:
If you want to train them with the cameras active you can build walls around each of your environments so that the drones can't see each other. 

Another approach is to train one model that can teach the drone to pass through given waypoints using Isaac Lab. then you can use the Isaac Lab replays to capture annotated data from a single environment and use that data set to train a model to generate waypoints from a camera feed. This is the approach at the moment and I believe it will be a good one.

Bruce-Si, 2025-03-21:
> I need to use camera data for training in Isaac Lab, such as in drone navigation. I have added a camera to each drone in parallel environments, but I want to ensure that drones in one environment [...]

+1.
I need the same functionality, but it doesn't seem possible to achieve this in visual simulation.

Fear113, 2025-03-22:
> I need to use camera data for training in Isaac Lab, such as in drone navigation. I have added a camera to each drone in parallel environments, but I want to ensure that drones in one environment [...]

Have you solved your problem? I have a similar problem and don't know how to solve it either

Atticlmr, 2025-03-26:
> Have you solved your problem? I have a similar problem and don't know how to solve it either

Simply ensure that each environment is spaced far enough apart so that the cameras cannot see each other.

closed 2025-04-15 by RandomOakForest