isaac-sim/IsaacLab #2103: [Question] efffort clipping in DCMotor

IoannisDadiotis, 2025-03-18:
Hi,

In `DCMotor` actuator class the efforts are clipped based on the following function (defined [here](https://github.com/isaac-sim/IsaacLab/blob/91f53e2fe5702eb2fd4bc4de98ab19d320f772c9/source/isaaclab/isaaclab/actuators/actuator_pd.py#L276) )

```python
    def _clip_effort(self, effort: torch.Tensor) -> torch.Tensor:
        # compute torque limits
        # -- max limit
        max_effort = self._saturation_effort * (1.0 - self._joint_vel / self.velocity_limit)
        max_effort = torch.clip(max_effort, min=self._zeros_effort, max=self.effort_limit)
        # -- min limit
        min_effort = self._saturation_effort * (-1.0 - self._joint_vel / self.velocity_limit)
        min_effort = torch.clip(min_effort, min=-self.effort_limit, max=self._zeros_effort)

        # clip the torques based on the motor limits
        return torch.clip(effort, min=min_effort, max=max_effort)
```

As can be seen, the effort is clipped using the `effort_limit` value and not the `saturation_effort` (which is higher compared to `effort_limit`). Even if we consider a joint velocity in the vicinity of zero, the applied effort can never reach the saturation values as it never exceeds the `effort_limit`. Therefore, I was wondering what is the logic behind this clipping and why there is a `saturation_effort` value if this value is never reached? 

Thank you in advance,
Ioannis

Mayankm96, 2025-03-19:
The `saturation_limit` is the peak motor effort limit while the `effort_limit` within which you want to operate the motor. Thus, the two are kept as separated concepts.

IoannisDadiotis, 2025-03-19:
Isn't it too conservative to clip the effort to `effort_limit`? Given that a physical motor can reach `saturation_limit` for short periods.

Mayankm96, 2025-03-19:
This depends on your application and how much current supply you can provide to the robot. For robots we work with in our lab, the on-board battery can only provide 8A (40N-m) meanwhile the actual drive can take up to 30A (~100N-m). So the saturation limit follows what the drive does and the effort limit is what is realistically possible to actuate given the hardware consideration.

If in your case, you can operate the joint at the full capacity of the drive, you can set the effort limit to be high. In some cases (like the Spot robot), there is a customized clipping scheme so they et the effort limit to infinity.

IoannisDadiotis, 2025-03-20:
This is clear, thanks

closed 2025-03-20 by IoannisDadiotis