isaac-sim/IsaacLab #2110: failed to find the package that core 16 belongs to.

Yandong-Luo, 2025-03-18:
[image]

When I install isaac-sim I got the error about "failed to find the package that core 16 belongs to." But the selector can works

[image]

RandomOakForest, 2025-03-19:
Thanks for posting this. How did you install Isaac Sim? Are you following [the steps to install Isaac Lab and Isaac Sim in the docs here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html)? If you can elaborate on how are you using environments and installing the tools would help.

DawnyWu, 2025-03-21:
same problem

Yandong-Luo, 2025-03-22:
> Thanks for posting this. How did you install Isaac Sim? Are you following [the steps to install Isaac Lab and Isaac Sim in the docs [...]

I strictly followed all the installation requirements. My system is Ubuntu 20.04 and the graphics card driver is 535.230.02

YuHoChau, 2025-03-26:
Same error here

kisaramei, 2025-03-28:
I have the same problem

RandomOakForest, 2025-04-03:
Thanks for following up. Something may have changed that affects the Isaac Sim selector script. The team is reviewing this. Thanks.

RandomOakForest, 2025-04-04:
Following up on this, are you running inside a docker container? could you share the output of nvidia-smi? Thanks.

YuHoChau, 2025-04-15:
FYI. Thank you.
[image]

fansihub, 2025-06-28:
I have the same problem

kellyguo11, 2025-07-07:
Could you try a fresh install as recommended by https://github.com/isaac-sim/IsaacSim/issues/29?

AntoineRichard, 2026-08-27:
@Yandong-Luo thanks for flagging this, and to everyone who chimed in with the same message.

The warning comes from the Isaac Sim 4.5 workstation selector script, not from any Isaac Lab code — there's no reference to it anywhere in this repo. It was tracked upstream as https://github.com/isaac-sim/IsaacSim/issues/29, which has since been closed. On top of that, Isaac Lab 3.x installs Isaac Sim through the pip/uv packages, so the selector isn't part of the supported setup path at all.

Closing here; please reopen if you see it on a supported 3.x install.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard