isaac-sim/IsaacLab #2115: [Question] Custom UR3 with Robotiq 2f-140 instability

ZachAtkins0n, 2025-03-19:
I'm using a custom usd file with a UR3 robot arm and Robotiq 2F-140 gripper using the assembler function in IsaacSim. 
When I use this for any reinforcement learning tasks with the robot is unstable which causes the value function loss to exponentially grow. 
I'm not sure if it is because of the usd file itself or something else. Any help will be appreciated.

Link to usd file:
https://drive.google.com/drive/folders/1chWPzIpIeduMP-0AUxyvLv0MNCVKrktF?usp=sharing

ZachAtkins0n, 2025-03-20:
I've modified the positions of the robot arm to reduce clipping however joint vel fluctuates with the reward alternating between a small number or exponentially tiny as shown in the image. I'm using the manager based lift environment using rsl_rl with the custom UR3 robot

[image]

RandomOakForest, 2025-03-20:
Thank you for posting this. I'll move the post to our _Discussions_ section for the team to follow up.