isaac-sim/IsaacLab #2122: [Bug Report] Lidar sensor data is abnormal when using the class 'DirectRLEnv'

YT-wk, 2025-03-20:
### Describe the bug

I tried to use `skrl` to perform reinforcement learning training in IsaacLab. I used Direct Workflow to build my reinforcement learning environment.
The LiDAR sensor data is only available in environment 0, and even then, the ray casting does not originate from the correct prim path location specified for the robot.

### Steps to reproduce

1. The code involved in the lidar operation is as follows:
```python
@configclass
class LocalNavEnvCfg(DirectRLEnvCfg):
    ray_caster = RayCasterCfg(
        prim_path="/World/envs/env_.*/Robot/base_link/lidar_cage",
        update_period=1 / 60,      
        offset=RayCasterCfg.OffsetCfg(pos=(0., 0., 0.)),   
        mesh_prim_paths=["/World/envs"],
        attach_yaw_only=True,          

        pattern_cfg=patterns.LidarPatternCfg(
            channels=100, vertical_fov_range=[-45., 45.], horizontal_fov_range=[-90, 90], horizontal_res=1.0      
        ),
        debug_vis=True,
        max_distance=laser_max_range 
    )
    def _setup_scene(self):
        self._ray_caster = self.cfg.ray_caster.class_type(self.cfg.ray_caster)
        self.scene.sensors["ray_caster"] = self._ray_caster

    def _simulate_laser_scan(self) -> torch.Tensor:
        env_origins = self.scene.env_origins
        env_ids = torch.arange(self.num_envs, dtype=torch.int, device=self.device)
        num_envs = len(env_ids)
        num_rays = self.scene["ray_caster"].data.ray_hits_w.shape[1]
        print("ray_caster_data is ", self.scene["ray_caster"].data.ray_hits_w)

        robot_rot_inv, robot_pos_inv = tf_inverse(self.robot_true_position[:, 3:7], self.robot_true_position[:, :3])

        local_rot = torch.tensor([1.0, 0.0, 0.0, 0.0], device=self.device).repeat(num_envs, 1)
        local_rot = local_rot.unsqueeze(1)
        local_rot = local_rot.repeat_interleave(num_rays, dim=1)

        hits_local_rot, hits_local_pos = tf_combine(
            robot_rot_inv.unsqueeze(1).repeat_interleave(num_rays, dim=1),
            robot_pos_inv.unsqueeze(1).repeat_interleave(num_rays, dim=1),
            local_rot,
            self.scene["ray_caster"].data.ray_hits_w - env_origins.unsqueeze(1)
        )

```
2. Create multiple environments with cloned robots
3. Try to access LiDAR data across environments
4. Observe data only exists for env 0 and rays don't originate from robot's lidar_cage
```
ray_caster_data is  tensor([[[ 30.0000, -29.7786,   0.2521],
         [ 30.0014, -29.7786,   0.2521],
         [ 30.0027, -29.7785,   0.2521],
         ...,
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf]],

        [[     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         ...,
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf]],

        [[     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         ...,
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf]],

        ...,

        [[     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         ...,
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf]],

        [[     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         ...,
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf]],

        [[     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         ...,
         [     inf,      inf,      inf],
         [     inf,      inf,      inf],
         [     inf,      inf,      inf]]], device='cuda:0')
```
The red part of the robot in the figure below is the lidar point visualization:
[image]

When it works correctly, it should be as shown in the figure below:
[image]

### System Info

• Commit: [IsaacLab commit hash]
• Isaac Sim Version: [4.5.0]
• Isaac Lab Version: [2.0.0]
• OS: [Ubuntu 20.04]
• GPU: [RTX 4080]
• CUDA: [12.8.61]
• GPU Driver: [535.183.01]

### Additional context

The issue occurs when using wildcards in prim_path for multi-environment setup. The ray caster seems to only bind to the first matching prim path instead of creating per-environment sensors. Additionally, ray directions don't respect the specified lidar_cage transform.

### Checklist

- [x] Checked for similar issues
- [x] Verified issue is specific to Isaac Lab

RandomOakForest, 2025-03-21:
Thank you for posting this. The team will investigate.

AntoineRichard, 2026-08-27:
Thanks @YT-wk for the detailed report and the side-by-side visualizations — they made it easy to pin down what was happening.

The behavior you saw is a documented limitation of the original `RayCaster`: it supports only a single static mesh target. With `mesh_prim_paths=["/World/envs"]` it binds to the first matching mesh it finds, so rays only ever hit geometry near env 0 and every other environment returns `inf`. The docstring in `source/isaaclab/isaaclab/sensors/ray_caster/ray_caster_cfg.py` still states this restriction explicitly.

The good news: this use case is now covered by the newer **`MultiMeshRayCaster`** sensor (added in https://github.com/isaac-sim/IsaacLab/pull/3298, available in `isaaclab.sensors`). It resolves ray-cast targets per environment and can track moving meshes. For your setup it would look like:

```python
from isaaclab.sensors import MultiMeshRayCasterCfg

ray_caster = MultiMeshRayCasterCfg(
    prim_path="/World/envs/env_.*/Robot/base_link/lidar_cage",
    mesh_prim_paths=[
        MultiMeshRayCasterCfg.RaycastTargetCfg(
            prim_expr="{ENV_REGEX_NS}/<your_target>",  # per-env targets, e.g. obstacles
            track_mesh_transforms=True,  # keeps up with meshes that move
        ),
        "/World/ground",  # plain strings still work for static meshes
    ],
    pattern_cfg=...,
)
```

`MultiMeshRayCasterCfg` lives in `source/isaaclab/isaaclab/sensors/ray_caster/multi_mesh_ray_caster_cfg.py`; a worked multi-target example is in the class docstring of `source/isaaclab/isaaclab/sensors/ray_caster/base_multi_mesh_ray_caster.py`. One more note for migration: `attach_yaw_only` has been replaced by `ray_alignment` — for a 3D lidar pattern like yours you likely want `ray_alignment="base"` so the rays follow the full sensor orientation.

Since the multi-environment, per-robot ray-casting case is now supported by `MultiMeshRayCaster` on a current release, I'll close this one. If you still see missing hits or wrong ray origins with the new sensor on a recent Isaac Lab, please reopen — happy to dig in.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard