isaac-sim/IsaacLab #2127: [Question] What exactly is going on with force/torque sensors

ParlitsisG, 2025-03-21:
### Question
I am trying for the past month to find a solution to create a force torque sensor for an articulation yet with no success.
The main issue is this question/issue has already been adressed discussed in different questions but due to the different versions of Isaac lab and Issac sim is hard to follow what exactly applies and whether there is a method to get either 6d measured wrench in a body or whether someone create torque sensors on joints(or use something equivalent)
Relative Issues:
#1725 
#139
#1202
#1092

Any feedback with any type of implementation that can actually work is more than welcome its just that certain issues of this matter are either unanswered or part of the explanation is not applicable with the latest Isaac lab and Isaac sim versions.

RandomOakForest, 2025-03-23:
Thank you for posting this. The team is working on enhancements in this regard. I'll review with the team and we'll try to post a more definitive clarification here. Thank you for the detailed review of past issues on this.

songwookim, 2025-04-09:
robot_core.get_measured_joint_forces()  # robot_core : Articulation 
robot_core.get_measured_joint_efforts()

Is this any different from using above functions to get a force/torque value?

jtigue-bdai, 2025-04-09:
> robot_core.get_measured_joint_forces() # robot_core : Articulation robot_core.get_measured_joint_efforts()
> 
> Is this any different from using above functions to get a force/torque value?
@songwookim

These Articulation API calls are throught he isaacsim.core.articualtions. They are a slightly different API that the Articulations in IsaacLab.

The new API being added in #2128 provides the `ArticulationData.body_incoming_joint_wrench_b` that will provide the 6D F/T measurements at the proximal joint of each body in an articulation.

songwookim, 2025-04-11:
@jtigue-bdai  Thanks for the reply! I’m looking forward to the update.

I’m also working on variable impedance controller tasks in Isaac Sim, and your contributions have been sincerely helpful to my work :)

closed 2025-04-16 by RandomOakForest