isaac-sim/IsaacLab #2130: Added joints not recognized

omarrayyann, 2025-03-22:
I keep getting this error despite properly adding the joint into my .usd
```
ValueError: Not all regular expressions are matched! Please check that the regular expressions are correct: 
        D6Joint: []
Available strings: []
```

RandomOakForest, 2025-03-23:
Thanks for posting this. Which version of Isaac Lab and Isaac Sim are you using?

omarrayyann, 2025-03-24:
I tried adding revolute, it also did not work.

shafeef901, 2025-03-26:
If you had converted from a URDF to USD first, and the joint is a fixed joint, make sure to add `dont_collapse="true"`. Something like: `<joint name="index_joint_4" type="fixed" dont_collapse="true">`

omarrayyann, 2025-03-26:
Hello @shafeef901. I was adding it on isaacsim directly and saving it as .USD directly. I did not import external URDFs

RandomOakForest, 2025-04-16:
Following up, if you still face this issues please open a _bug report_, and try Isaac Lab 2.0.2 or later. I will close this issue for now. Thank you for your interest in Isaac Lab.

closed 2025-04-16 by RandomOakForest