isaac-sim/IsaacLab #2141: The 5070TI is not functioning properly with IsaacLab

npcliu, 2025-03-25:
As is widely known, the 5070TI requires the installation of the PyTorch development version (CUDA 12.8). I followed the tutorial to install IsaacLab, then upgraded PyTorch to the dev version. However, the installation process reported an error indicating that the current IsaacLab version is incompatible with the PyTorch dev version:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
isaaclab 0.36.3 requires torch==2.5.1, but you have torch 2.8.0.dev20250324+cu128 which is incompatible.
isaaclab-rl 0.1.1 requires torch==2.5.1, but you have torch 2.8.0.dev20250324+cu128 which is incompatible.
isaaclab-tasks 0.10.26 requires torch==2.5.1, but you have torch 2.8.0.dev20250324+cu128 which is incompatible.
torchvision 0.20.1 requires torch==2.5.1, but you have torch 2.8.0.dev20250324+cu128 which is incompatible.
```

When I followed the tutorial to run the command `./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py `to launch and test IsaacLab, the terminal displayed the following error:

```
2025-03-25 02:47:40 [8,941ms] [Error] [omni.ext._impl._internal] Failed to import python module isaaclab_tasks from /home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks. Error: operator torchvision::nms does not exist. Traceback:
Traceback (most recent call last):
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 187, in _custom_importer
    return _import_public(ext_module.path, ext_module.name, reload_enabled)
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py", line 94, in _import_public
    module = import_module(module_name)
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/__init__.py", line 30, in <module>
    import_packages(__name__, _BLACKLIST_PKGS)
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/importer.py", line 40, in import_packages
    for _ in _walk_packages(package.__path__, package.__name__ + ".", blacklist_pkgs=blacklist_pkgs):
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/importer.py", line 87, in _walk_packages
    yield from _walk_packages(path, info.name + ".", onerror, blacklist_pkgs)
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/importer.py", line 87, in _walk_packages
    yield from _walk_packages(path, info.name + ".", onerror, blacklist_pkgs)
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/importer.py", line 87, in _walk_packages
    yield from _walk_packages(path, info.name + ".", onerror, blacklist_pkgs)
  [Previous line repeated 1 more time]
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/importer.py", line 75, in _walk_packages
    __import__(info.name)
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/stack/config/franka/__init__.py", line 8, in <module>
    from . import (
  File "/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/stack/config/franka/stack_ik_rel_blueprint_env_cfg.py", line 8, in <module>
    from torchvision.utils import save_image
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/torchvision/__init__.py", line 10, in <module>
    from torchvision import _meta_registrations, datasets, io, models, ops, transforms, utils  # usort:skip
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/torchvision/_meta_registrations.py", line 164, in <module>
    def meta_nms(dets, scores, iou_threshold):
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/torch/library.py", line 1023, in register
    use_lib._register_fake(op_name, func, _stacklevel=stacklevel + 1)
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/torch/library.py", line 214, in _register_fake
    handle = entry.fake_impl.register(func_to_register, source)
  File "/home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/torch/_library/fake_impl.py", line 31, in register
    if torch._C._dispatch_has_kernel_for_dispatch_key(self.qualname, "Meta"):
RuntimeError: operator torchvision::nms does not exist

2025-03-25 02:47:40 [8,941ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaaclab_tasks' in '/home/liu/bruce/legged_lab/IsaacLab/source/isaaclab_tasks' failed to load.

At:
  /home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py(213): startup
  /home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/omni/kernel/py/omni/ext/_impl/_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  /home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(427): _start_app
  /home/liu/anaconda3/envs/leggedlab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(212): __init__
  /home/liu/bruce/legged_lab/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py(642): _create_app
  /home/liu/bruce/legged_lab/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py(114): __init__
  /home/liu/bruce/legged_lab/IsaacLab/scripts/tutorials/00_sim/create_empty.py(29): <module>

2025-03-25 02:47:40 [8,941ms] [Error] [omni.ext.plugin] [ext: isaaclab_tasks-0.10.26] Failed to startup python extension.
```
How can I resolve this issue?

Daimy0109, 2025-04-06:
I had a similar problem with my 5080, I found that every time I executed the shell “isaaclab.bat --install” it overwrote my previous version of torch, so after I executed the isaaclab.bat --install command again, I went to pytorch https://download .pytorch.org/whl/nightly/cu128 and found the corresponding torchvision and torch, and downloaded them directly.

Daimy0109, 2025-04-06:
But every time I run a sample script, such as launch_app.py, he gets a similar warning “2025-04-06 08:20:33 [6,565ms] [Warning] [rtx.neuraylib.plugin] [IRAY:RENDER] 1.1 IRAY   rend warn : GPU 1 (NVIDIA GeForce RTX 5080) with CUDA compute capability 12.0 is unsupported by this version of iray photoreal. please look for an update please look for an update with your software provider.” I don't know what this will do!

Stefano-retinize, 2025-04-06:
> I had a similar problem with my 5080, I found that every time I executed the shell “isaaclab.bat --install” it overwrote my previous version of torch, so after I executed the isaaclab.bat --install [...]

Hey Thanks for this information! DId you download the .whl from that link? What did you do with the .whl? Thanks!

closed 2025-04-16 by RandomOakForest

ayaancdry, 2025-06-28:
> > I had a similar problem with my 5080, I found that every time I executed the shell “isaaclab.bat --install” it overwrote my previous version of torch, so after I executed the isaaclab.bat --install command again, I went to pytorch https://download .pytorch.org/whl/nightly/cu128 and found the corresponding torchvision and torch, and downloaded them directly.
> 
> Hey Thanks for this information!
> **I can confirm that this way around worked for me!.**
> The key though is to run both: 
> `pip install torch-2.7.0.dev20250312+cu128-cp310-cp310-win_amd64.whl --no-deps`
> and 
> `pip install torchvision-0.22.0.dev20250405+cu128-cp310-cp310-win_amd64.whl --no-deps`
> 
> So you don't force torch to change the other dependencies that don't match with the isaaclab version.

So are you able to use isaacsim properly w/o any issues after doing this procedure?