isaac-sim/IsaacLab #2145: [Question] [Error] [carb.launcher.plugin]

Huisouan, 2025-03-25:
I got this error everytime when launching the isaacsim simulator.Does amyone know how to fix this problem?

[image]

`
[51.733s] app ready
2025-03-25 08:24:26 [51,759ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7374781440
2025-03-25 08:24:26 [51,759ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false
2025-03-25 08:24:26 [51,759ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7301033856
2025-03-25 08:24:26 [51,759ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : New limit 9748724 (slope: 439, intercept: 13179904)
2025-03-25 08:24:26 [51,760ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 4287352704
2025-03-25 08:24:26 [51,760ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-03-25 08:24:27 [52,304ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin] [parent]: timed out waiting for the child process 132787 to exec {errno = -1/Unknown error -1}
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin] failed to fork the child process {errno = -1/Unknown error -1, lastErrno = 11/Resource temporarily unavailable}
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin] launch descriptor for '/home/hsh/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni.telemetry.transmitter/omni.telemetry.transmitter' (54 arguments)
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]     cmd: /home/hsh/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni.telemetry.transmitter/omni.telemetry.transmitter --/telemetry/enableAnonymousAppName=true --/telemetry/enableAnonymousData=true --/telemetry/useOpenEndpoint=true "--/telemetry/mode=""" "--/telemetry/logging/warnFilter/0="*"" --/telemetry/enableNVDF=false --/telemetry/enableSentry=false --/telemetry/removePIISentry=false "--/telemetry/appRootPath=""" "--/telemetry/dsn="https://b61f14dfc36f4beabadf4d9dba8e492b@sentry.perflab.nvidia.com/834"" "--/telemetry/openEndpointUrl="https://opentelemetry.analytics.nvidiagrid.net/api/v3/topic/omniverse-team.open.omni-open-collector"" "--/telemetry/openTestEndpointUrl="https://opentelemetry.analytics.nvidiagrid.net/api/v3/topic/omniverse-team.open.omni-test-open-collector"" "--/telemetry/enterpriseOpenEndpointUrl="https://opentelemetry.analytics.nvidiagrid.net/api/v3/topic/omniverse-team.open.omni-ove-open-collector"" "--/telemetry/enterpriseOpenTestEndpointUrl="https://opentelemetry.analytics.nvidiagrid.net/api/v3/topic/omniverse-team.open.omni-ove-test-open-collector"" "--/log/outputStreamLevel="Warning"" "--/log/debugConsoleLevel="Warning"" "--/log/fileLogLevel="Info"" --/log/fileMaxCount=30 --/log/includeTimeStamp=true "--/log/setElapsedTimeUnits="milliseconds"" --/log/enabled=true --/log/async=false "--/log/fileFlushLevel="verbose"" --/log/flushStandardStreamOutput=false --/log/enableDebugConsoleOutput=true --/log/enableColorOutput=true --/log/processGroupId=0 --/log/includeSource=true --/log/includeChannel=true --/log/includeFilename=false --/log/includeLineNumber=false --/log/includeFunctionName=false --/log/includeThreadId=false --/log/includeProcessId=false --/log/detail=false --/log/fullDetail=false --/log/forceAnsiColor=false --/log/fileAppend=false --/log/enableStandardStreamOutput=false --/log/file=/home/hsh/.nvidia-omniverse/logs/omni.telemetry.transmitter.log --/log/level=Info --/log/file=/home/hsh/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/logs/omni.telemetry.transmitter.log --/log/level=Warning --/privacy/personalization=true --/privacy/performance=true --/privacy/usage=true --/privacy/userId=6e7edda4245b6e32 --/telemetry/extraFieldsToAdd=schemaid --/structuredLog/extraFields/schemaid=omniverse-team.open.omni-open-collector:1 --/telemetry/runEnvironment=Individual --/telemetry/transmitter/0/authenticate=false --/telemetry/transmitter/0/eventProtocol=defaultWithList --/telemetry/transmitter/0/endpoint=https://opentelemetry.analytics.nvidiagrid.net/api/v3/topic/omniverse-team.open.omni-open-collector
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]     flags = NoInheritEnv, AllowBadEnv, NoStdOut, NoStdErr (0x000005b0)
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]     env:
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       0) '__GL_0x50a159=0x00000000'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       1) 'NODEVICE_SELECT=1'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       2) 'VULKAN_HEADERS_INSTALL_DIR='
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       3) 'VULKAN_SDK='
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       4) 'VK_SDK_PATH='
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       5) 'ISAAC_PATH=/home/hsh/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       6) 'EXP_PATH=/home/hsh/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/apps'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       7) 'TERM_PROGRAM=vscode'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       8) 'GIO_LAUNCHED_DESKTOP_FILE=/usr/share/applications/code.desktop'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       9) 'DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       10) 'ORIGINAL_XDG_CURRENT_DESKTOP=ubuntu:GNOME'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       11) 'GDMSESSION=ubuntu'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       12) 'PATH=/usr/local/cuda-12.6/bin:/home/hsh/miniconda3/bin:/home/hsh/miniconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/hsh/.vscode/extensions/ms-python.debugpy-2025.4.1-linux-x64/bundled/scripts/noConfigScripts'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       13) 'JOURNAL_STREAM=9:20432'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       14) 'VSCODE_GIT_ASKPASS_NODE=/usr/share/code/code'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       15) 'XDG_SESSION_TYPE=wayland'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       16) 'HOME=/home/hsh'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       17) 'LOGNAME=hsh'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       18) 'CONDA_PREFIX=/home/hsh/miniconda3'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       19) 'XDG_SESSION_DESKTOP=ubuntu'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       20) 'BUNDLED_DEBUGPY_PATH=/home/hsh/.vscode/extensions/ms-python.debugpy-2025.4.1-linux-x64/bundled/libs/debugpy'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       21) 'SESSION_MANAGER=local/hsh-System-Product-Name:@/tmp/.ICE-unix/2531,unix/hsh-System-Product-Name:/tmp/.ICE-unix/2531'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       22) 'GTK_MODULES=gail:atk-bridge'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       23) 'GJS_DEBUG_OUTPUT=stderr'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       24) 'SYSTEMD_EXEC_PID=2584'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       25) 'VSCODE_GIT_ASKPASS_MAIN=/usr/share/code/resources/app/extensions/git/dist/askpass-main.js'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       26) 'GJS_DEBUG_TOPICS=JS ERROR;JS LOG'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       27) 'PWD=/home/hsh/Code/sim_assets'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       28) 'PYDEVD_DISABLE_FILE_VALIDATION=1'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       29) 'SHELL=/bin/bash'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       30) 'QT_ACCESSIBILITY=1'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       31) 'XDG_MENU_PREFIX=gnome-'
2025-03-25 08:24:32 [57,383ms] [Error] [carb.launcher.plugin]       32) 'COLORTERM=truecolor'
[... cut ...]
2025-03-25 08:24:32 [57,384ms] [Error] [carb.launcher.plugin]       80) 'USERNAME=hsh'
2025-03-25 08:24:32 [57,384ms] [Error] [carb.launcher.plugin]       81) 'MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA='
2025-03-25 08:24:32 [57,384ms] [Error] [carb.launcher.plugin]       82) 'CONDA_DEFAULT_ENV=base'
2025-03-25 08:24:32 [57,384ms] [Error] [carb.launcher.plugin]       83) 'DEBUGINFOD_URLS=https://debuginfod.ubuntu.com '
2025-03-25 08:24:32 [57,384ms] [Warning] [carb.launcher.plugin] failed to launch the new process.
`

RandomOakForest, 2025-04-01:
Thanks for posting this. How are you launching Isaac Sim? How did you install Isaac Lab and Isaac Sim?

Huisouan, 2025-04-02:
By following the document below.
`
https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html
`

RandomOakForest, 2025-04-16:
Following up, it seems a possibility is that the system has run out of process IDs, which is possible if there are a lot of processes in your system. Have you tried restarting and trying again?