isaac-sim/IsaacLab #2146: [Question] Lula Kinematics and Isaac Lab

btabia, 2025-03-25:
Hello, 

I want to reuse my Lula kinematics that I was using in Isaac Sim to Isaac Lab 2.0. 

It looks that SingleArticulation does not work with Isaac Lab. Does there is an incompatibility? 

Many Thanks

Here the error that I've got: 

```
  File "/home/train_tqc.py", line 130, in main
    env = gymnasium.make(id="BaseEnv-v0", cfg_env = wandb.config["env"], cfg_task  = wandb.config["tasks"])
  File "/home/anaconda3/envs/il/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 740, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/base_env.py", line 116, in __init__
    self.single_art.initialize()
  File "/home/anaconda3/envs/il/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.prims/isaacsim/core/prims/impl/single_articulation.py", line 308, in initialize
    self._articulation_view.initialize(physics_sim_view=physics_sim_view)
  File "/homeanaconda3/envs/il/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.prims/isaacsim/core/prims/impl/articulation.py", line 5128, in initialize
    self._on_physics_ready(None)
  File "/home/anaconda3/envs/il/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.prims/isaacsim/core/prims/impl/articulation.py", line 5134, in _on_physics_ready
    self._physics_view = simulation_view.create_articulation_view(
AttributeError: 'NoneType' object has no attribute 'create_articulation_view' 
```

RandomOakForest, 2025-04-01:
Thanks for posting this. Some classes were renamed in the release. You may have to edit your code a bit. Please review the [migration notes here](https://isaac-sim.github.io/IsaacLab/main/source/refs/migration.html#renaming-of-omni-isaac-core-classes).

btabia, 2025-04-09:
Hello, 

My question is simple, is Lula Kinematics from Isaac Sim usable and compatible in Isaac Lab?

The diff IK from Isaac Lab cause me troubles and I want to reuse the IK from Lula kinematics. 

Many Thanks

RandomOakForest, 2025-04-16:
Following up, `SingleArticulation` should work, probably it's just not being initialized in the right place. It needs to be initialized after simulation starts.

closed 2025-04-18 by RandomOakForest