isaac-sim/IsaacLab #2147: [Bug Report] XML to USD possible load bug

giulioturrisi, 2025-03-25:
### Describe the bug

I'm trying to convert the following XML (https://github.com/iit-DLSLab/gym-quadruped/blob/master/gym_quadruped/robot_model/hyqreal/hyqreal.xml) in USD. With previous IsaacLab version, i was able to convert different model inside the same repo, but now, i obtain the following structure

[image]

As you can see, the "base" is duplicated, whereas with the old IsaacLab version i was able to convert the model in a structure like this:

[image]

Due to this difference, i cannot load correctly the usd in IsaacLab, since i obtain the following error:

AttributeError: 'Articulation' object has no attribute '_root_physx_view'

### System Info

- Commit: c4bec8f

Thanks

Ycyofmine, 2025-04-07:
Solution for Ubuntu 22.04 users:
1. Delete the entire ov folder located at ~/.local/share/ov.  
2. Re-run the Isaac Sim selector script:  
```
./isaacsim-selector.sh
```
This should resolve the URDF import issue.

yash-s20, 2025-04-12:
deleting worldBody from the USD fixes the error you're seeing (`no attribute _root_physx_view`) @giulioturrisi . it happens because the USD has multiple root prims (also discussed [here](https://forums.developer.nvidia.com/t/error-please-ensure-that-there-is-only-one-articulation-in-the-prim-path-tree/311183))

However I'm having issues loading multiple assets where only one responds to joint pose changes. Is this issue related @RandomOakForest ?

https://github.com/user-attachments/assets/a4d57f5c-08b0-4839-8295-c6260c5eaa1f

RandomOakForest, 2025-07-03:
Thank you for following up. We'll review and aim to resolve this soon.

ooctipus, 2025-07-10:
xml to usd is currently a bit buggy, and urdf importer seems a lot more stabler, if it is possible, can you try to use urdf importer with isaaclab 2.1 + isaacsim 4.5 and see if it works?

louislelay, 2025-07-11:
Hey @giulioturrisi, @yash-s20, and @ooctipus, I’ve just submitted a PR to the Isaac Sim MJCF importer extension that fixes this issue. Feel free to take a look and give it a try!

giulioturrisi, 2025-07-11:
Great! I cannot find the link of the PR. Can you share it?

louislelay, 2025-07-11:
It appears in the conversation just before my last comment, but here's the link anyway: https://github.com/isaac-sim/mjcf-importer-extension/pull/9

Kenjiro-Minamikawa, 2025-07-15:
> Hey [@giulioturrisi](https://github.com/giulioturrisi), [@yash-s20](https://github.com/yash-s20), and [@ooctipus](https://github.com/ooctipus), I’ve just submitted a PR to the Isaac Sim MJCF importer extension that fixes this issue. Feel free to take a look and give it a try!

Hi, I used this tool and it remove the worldBody. But the problem @yash-s20 mentioned is still remained.
Are there any idea to solve this problem?

louislelay, 2025-07-30:
Hi @ooctipus, if the bugs are going to be fixed in a future version of Isaac Sim does my PR to the MJCF importer extension still make sense ?

jgillick, 2025-08-06:
FYI, [my comment](https://github.com/isaac-sim/IsaacLab/issues/2147#issuecomment-3092633674) seems to be wrong. My joints weren't moving because of another issue with my model.  My [original workaround](https://github.com/isaac-sim/IsaacLab/issues/2147#issuecomment-3072932364) is still valid, though.

giulioturrisi, 2025-08-09:
Hello @ooctipus, i still have the problem describe in the first post here with IsaacLab v2.2.0 and IsaacSim5.0

adrialopezescoriza, 2026-02-03:
Hello, I am using IsaacLab 5.1 and still experiencing the issue described here. Has a workaround been found?