isaac-sim/IsaacLab #2148: [Bug Report] Unable to force the training to stay only on one GPU

GiulioRomualdi, 2025-03-25:
### Describe the bug

I'm trying to run a training script by specifying the GPU I would like to use. However, all GPUs on the machine are being used, even when I request only one.

### Steps to reproduce

Run the training with:

```
python train.py --task Isaac-Cartpole-v0 --num_envs 32 --headless --device cuda:1
```

Then check `nvidia-smi`.

Before running the training, if I run `nvidia-smi`, I get the following. Since this is a shared machine, I cannot currently provide a situation where all GPUs are idle:

```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.107.02             Driver Version: 550.107.02     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA L40S                    Off |   00000000:21:00.0 Off |                    0 |
| N/A   46C    P0            114W /  350W |   18748MiB /  46068MiB |     36%      Default |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA L40S                    Off |   00000000:C1:00.0 Off |                    0 |
| N/A   30C    P8             32W /  350W |       2MiB /  46068MiB |      0%      Default |
+-----------------------------------------+------------------------+----------------------+
|   2  NVIDIA L40S                    Off |   00000000:E1:00.0 Off |                    0 |
| N/A   45C    P0            143W /  350W |   43829MiB /  46068MiB |      0%      Default |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|=========================================================================================|
|    0   N/A  N/A   1757715    C+G   ...b/_isaac_sim/kit/python/bin/python3      18707MiB |
|    2   N/A  N/A   2288867      C   python                                      43820MiB |
+-----------------------------------------------------------------------------------------+
```

However, after launching the command:
```
python train.py --task Isaac-Cartpole-v0 --num_envs 32 --headless --device cuda:1
```

I observe that GPU 0 and 2 are also slightly used, even though they weren't requested:

```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.107.02             Driver Version: 550.107.02     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA L40S                    Off |   00000000:21:00.0 Off |                    0 |
| N/A   46C    P0            118W /  350W |   19347MiB /  46068MiB |     63%      Default |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA L40S                    Off |   00000000:C1:00.0 Off |                    0 |
| N/A   32C    P0             82W /  350W |    2688MiB /  46068MiB |      8%      Default |
+-----------------------------------------+------------------------+----------------------+
|   2  NVIDIA L40S                    Off |   00000000:E1:00.0 Off |                    0 |
| N/A   45C    P0            106W /  350W |   44317MiB /  46068MiB |      0%      Default |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
[... 92 lines cut ...]
- Commit: https://github.com/isaac-sim/IsaacLab/tree/v2.0.2  
- Isaac Sim Version:  Isaac Sim 4.5
- OS: Ubuntu 20.04  
- GPU: NVIDIA L40S  
- CUDA: 12.4  
- GPU Driver: 550.107.02  

---

### Additional Context

---

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)  
- [x] I have verified that the issue is not caused by running Isaac Sim itself, but is related to this repository  

---

### Acceptance Criteria

Add the criteria for when this task is considered **done**. If not known at issue creation time, you can add it once the issue is assigned.

- [ ] Ensure that training runs **only** on the specified GPU and does not allocate memory or resources on other devices.

GiulioRomualdi, 2025-03-25:
cc @Giulero @traversaro

traversaro, 2025-03-25:
Can't we use the CUDA_VISIBLE_DEVICES env variable?

Mayankm96, 2025-03-25:
Which library are you using?

GiulioRomualdi, 2025-03-25:
> Which library are you using?
> 
> Some of them have a device argument that needs to be set explicitly through their configs.

I'm using the example script provided in isaaclab using the task mentioned here  https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/run_rl_training.html I use rsl_rl for ppo and the policy

GiulioRomualdi, 2025-03-25:
> Can't we use the CUDA_VISIBLE_DEVICES env variable?

Sure! I can try tomorrow morning

Mayankm96, 2025-03-25:
The device is set to "cuda:0" by default here:
> https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_rl/isaaclab_rl/rsl_rl/rl_cfg.py#L83-L84

If you check the train.py, it is not overwritten anywhere so the device is always "cuda:0" which leads to the behavior you see above:

> https://github.com/isaac-sim/IsaacLab/blob/main/scripts/reinforcement_learning/rsl_rl/train.py#L132

You can add an argument to train/play scripts to take learning agent device as additional input. Or simply pass the environment device to agent device as well.

closed 2025-04-16 by RandomOakForest