isaac-sim/IsaacLab #2155: [Bug Report] Odometry ros2 action graph publishes zeros in IsaacLab

swgart, 2025-03-25:
The odometry action graph does not update when using IsaacLab. It just publishes zeros. This issue does not occur in IsaacSim.

### Steps to Reproduce

    Run the odom_check branch of [GitHub - UWRobotLearning/GroundControl: The Isaac Sim Extension for Ground Autonomy](https://github.com/UWRobotLearning/GroundControl.git)
    python scripts/environments/teleoperation/teleop_se2_agent_ROS2.py --task Isaac-Navigation-Flat-Go2-Play-v0 --num_envs 1
    In a separate terminal run ros2 topic echo /go2/platform/odom --field pose.pose

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: RTX5000 ADA
- CUDA: 12.1
- GPU Driver: 535.183.01 

- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2025-03-25:
Please check the discussion here: https://github.com/isaac-sim/IsaacLab/issues/2078

swgart, 2025-03-26:
Thank you for the reference. Changing fabric and device settings did not seem to make a difference. Also for more information: the IMU action graph actually publishes everything except angular velocity, which stay zero. The LiDAR action graph appears to work in IsaacLab 2.0 but publishes blank messages in 1.2.

shiven2001, 2025-04-02:
Hi, I added a similar issue on this. I also get zeros. I think it is because IMU uses numpy and not tensor. Check this issue: https://github.com/isaac-sim/IsaacLab/issues/2089 

> Thank you for the reference. Changing fabric and device settings did not seem to make a difference. Also for more information: the IMU action graph actually publishes everything except angular [...]

closed 2025-04-16 by RandomOakForest