isaac-sim/IsaacLab #2165: [Question] How to completely uninstall isaac lab during version switching

MARMOTatZJU, 2025-03-26:
### Question

During switching between different version of Isaac Lab, I encountered errors with dependencies of Isaac Lab components, so I wish to uninstall my old version of Isaac Lab and reinstall it. But I did not find any command nor method to uninstall it. Is there any method to do it?

Thanks in advacne!

Mayankm96, 2025-03-26:
For uninstallation, there isn't alot you need to do from IsaacLab side. If you are using the same Isaac Sim version (which you probably are if you are using v2.X), then just simply switching to the tag should work. All python modules are installed in editable mode so the module code should get updated automatically.

> INFO: pip is looking at multiple versions of isaaclab-rl[all] to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement rsl-rl (unavailable) (from isaaclab-rl[all]) (from versions: none)
ERROR: No matching distribution found for rsl-rl (unavailable)

This was a bug because rsl-rl changed its package name to `rsl-rl-lib`. We did a minor patch fix for that in v2.0.1. Could you use that instead?

MARMOTatZJU, 2025-03-28:
@Mayankm96  Yeah I also found that the problem is fixed by version later than `v2.0.0`. I installed IsaacSim and IsaacLab via pip by following [official pip installation guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/isaaclab_pip_installation.html). This environment worked for me. IsaacLab 
 version is `v2.0.2` by my side.

However, I found another problem with the version of `rsl_rl_lib`. Its corresponding Python package `rsl_rl` can be imported by my side but no version is tagged, neither by running `python -c "import rsl_lib;print(rsl_lib.__version__)"` (which said rls_lib has no attribute `__version__`) or `pip list|grep rsl` (which returned an empty list). Do you have any idea with this problem?

Thanks in advance~

Mayankm96, 2025-04-02:
You can do something like this to get the version of rsl-rl-lib installed:

```python
import platform
from importlib.metadata import version

print(version("rsl-rl-lib"))

import rsl_rl
```

I think `rsl_rl` does not have the dunder for the version in its module. That's why you don't see that work. It's an issue on that side and not related to IsaacLab.

MARMOTatZJU, 2025-04-03:
@Mayankm96  Thanks for this advice! Besides, I also found that `pip show rsl-rl-lib` worked.

My question is answered. Thanks for your helps!

closed 2025-04-03 by MARMOTatZJU