isaac-sim/IsaacLab #2174: [Bug Report] Unitree Go2 training extremely slow on rough terrain

yang-zj1026, 2025-03-27:
### Describe the bug

When training RL locomotion policy of Unitree Go2 on rough terrains, the environment fps is significantly slow, around ~40k on 4090, while the environment fps of other robots like Unitree H1 is ~160k.

### Steps to reproduce

./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task Isaac-Velocity-Rough-Unitree-Go2-v0 --headless

### System Info

Describe the characteristic of your environment:

- Commit: https://github.com/isaac-sim/IsaacLab/commit/b5fa0eb031a2413c182eeb54fa3a9295e8fd867c
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: RTX 4090
- CUDA: 12.4
- GPU Driver: 565.57.01

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

LZJ910, 2025-03-28:
I'm encountering the same issue: Isaac Sim 4.2 runs relatively slow, while Isaac Sim 4.5 performs even slower.

Nate711, 2025-04-03:
I've also found that training on the rough terrain is incredibly slow, while flat terrain is fast.

Nate711, 2025-04-14:
@ubgk Thanks for the extra test. Are you saying that updating isaacsim from 4.2 to 4.5 reduced the training speed by 50% or that just updating the dependencies from what's required in 4.2 to what's required in 4.5 caused that slowdown?

RandomOakForest, 2025-04-16:
Thank you for following up. I will move this to our _Discussions_ section. Let us know if we can be of further help.