isaac-sim/IsaacLab #2188: [Bug Report] RTX 5090: Blurry, Pixelated and Noisy Rendering in RTX Real-Time Mode on Ubuntu 22.04 (Driver 570.133.07 / 570.124.04)

renhong-zhang, 2025-03-30:
### Describe the bug
GPU: 5090(This bug *relates to 50-series card*)
When using **RTX - Real Time** rendering mode in Isaac Sim on Ubuntu 22.04, the rendered robot appears **extremely blurry, pixelated, and noisy**. The issue presents as:

- The robot body has a **fuzzy, voxel-like appearance**  
- The shadow is **heavily pixelated and inaccurate**

This occurs with both default models and USD files I import. I have not modified any materials or shaders. Switching to Windows 11 (driver 572.83) on the same machine renders the scene correctly, indicating this issue is specific to **Ubuntu + 50-series GPU + RTX Real-Time mode**.

### Steps to reproduce

1. Launch Isaac Sim 4.5.0 on Ubuntu 22.04  
2. Load any default robot scene or asset (or import a USD model)  
3. Switch to **RTX - Real Time** mode  
4. Observe pixelated rendering of robot and shadows

Mode: **RTX - Real-Time**

[image]

Mode: **RTX - Real-Time 2.0**

[image]

Mode: **Interactive(Path Tracing)**

[image]

### System Info

- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: RTX 5090
- CUDA: 12.8
- GPU Driver: 570.133.07 and 570.124.04

### Additional context

- The same issue **does not occur** on Windows 11 (driver 572.83)  
- Please prioritize a fix or provide a workaround — thank you!

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)  
- [x] I have confirmed that the issue is not related to custom shaders or materials  

### Acceptance Criteria

- [ ] Robots and shadows are rendered clearly in RTX - Real Time mode on Ubuntu 22.04  
- [ ] Rendering matches Windows behavior using the same scene and model

RandomOakForest, 2025-04-01:
Thanks for posting this. The team will investigate.

Nate711, 2025-04-04:
I added the following cli args to rsl_rl/train.py and play.py so that path tracing is automatically used with low samples per frame. 

With 1 sample per pixel per frame it's just as fast as RTX-Real-Time for me. 

For those wondering why this works, these cli_args are passed to AppLauncher, which in turn passes the args to SimulationApp which supports these parameters https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.kit/docs/index.html 

```
parser.add_argument(
    "--renderer",
    type=str,
    default="PathTracing",
    choices=["RayTracedLighting", "PathTracing"],
    help="Renderer to use.",
)
parser.add_argument(
    "--samples_per_pixel_per_frame",
    type=int,
    default=1,
    help="Number of samples per pixel per frame.",
)
```

renhong-zhang, 2025-04-04:
I'm currently using RTX - Interactive mode as a workaround for simpler scenes, but it has poor visual quality and performance in indoor scenarios. According to a recent [NVIDIA Developer Forum update](https://forums.developer.nvidia.com/t/the-isaac-sim-gui-is-blurry/327759/14), the issue with RTX - Real-Time rendering stems from compatibility problems between NVIDIA Blackwell GPUs and the Kit version in Isaac Sim (4.5.0 and 4.2.0). As Isaac Sim is closed-source, we have to wait for an official update to address this.

zhaowy312, 2025-04-15:
@renhong-zhang My system configuration is the same as yours, and I recently tried to use isaac lab. 

 In the official documentation, they mentioned For 50 series GPUs,please use the latest PyTorch nightly build instead of PyTorch 2.5.1,which comes with Isaac Sim: pip install--upgrade--pre torch torchvision--index-url https://download.pytorch.org/whl/nightly/cu128.

. After executing./isaaclab.sh--install rl_games, I followed their instructions, but encountered version conflict:isaaclab 0.36.6 requires torch==2.5.1,but you have torch 2.8.0.dev20250414+cu128 which is incompatible.

isaaclab-rl 0.1.4 requires torch==2.5.1, but you have torch 2.8.0.dev20250414+cu128 which is incompatible.

isaaclab-tasks 0.10.27 requires torch==2.5.1, but you have torch 2.8.0.dev20250414+cu128 which is incompatible. 

 I don't know if you have encountered this problem, please give me some help, thank you very much.

ologandavid, 2025-04-15:
Curious if anyone else encountered this issue and had any solutions.

> [@renhong-zhang](https://github.com/renhong-zhang) My system configuration is the same as yours, and I recently tried to use isaac lab.
>
> In the official documentation, they mentioned For 50 series [...]

zhaowy312, 2025-04-16:
@Nate711 It works！Many thanks！

singh-jayant, 2025-05-29:
## Weird, but pleasantly surprising observations

### Observation 1
When I run IsaacSim with `isaac-sim.sh` or `isaac-sim.selector.sh`, I get blurry rendering in the viewport with Real-Time and Real-Time 2.0 rendering. Auto option on DLSS works better, so does DLAA, but all still noisy. Interactive mode works best, but incredibly slow.

[image]

[image]

[image]

[image]

[image]

### Observation 2
Replicator data generation also generates noisy RGB images, but depth images are perfectly clean

[image]

[image]

### Observation 3
BUT, when I use `scripts/tutorials/00_sim/set_rendering_mode.py` from `IsaacLab` to launch the exact same USD, the rendering works clean and fast, no issues at all. Not sure why or how but I ain't complaining.

[image]

atyagi702, 2025-05-30:
I am using the exact same setup as described in by OP but on Ubuntu 20.04 and everything seems to work fine

atyagi702, 2025-05-30:
I am using the exact same setup as described by OP but on Ubuntu 20.04 and it seems to get rid of the noise. But I am still facing the issue in Ubuntu 22.04.

snknitheesh, 2025-06-03:
I am using the same setup with isaac-4.5. Facing the same issue. I even installed the apt version required for torch and torchvision. 

[image]

Any help would be very good. I get the same output in interactive and real-time modes.

xuxuezhou, 2025-06-25:
I am facing the following error when I run scripts without headless. I have install the latest PyTorch nightly build. Wondering if anyone have the same issue and know how to solve it, thanks!

`RuntimeError: nvrtc: error: invalid value for --gpu-architecture (-arch)`

**System info**

- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: RTX 5090
- CUDA: 12.8
- GPU Driver: 570.153.02

Hahalim2022y, 2025-06-27:
> I am facing the following error when I run scripts without headless. I have install the latest PyTorch nightly build. Wondering if anyone have the same issue and know how to solve it, thanks! [...]

Hi, this is because cuda does not match the GPU. I used the following command to solve it. I hope it can help you!
$ pip list | grep -i cuda
nvidia-cuda-cupti-cu11       11.8.87
nvidia-cuda-cupti-cu12       12.8.90
nvidia-cuda-nvrtc-cu11       11.8.89
nvidia-cuda-nvrtc-cu12       12.8.93
nvidia-cuda-runtime-cu11     11.8.89
nvidia-cuda-runtime-cu12     12.8.90
nvidia-lula-no-cuda          0.10.1
$ pip uninstall -y nvidia-cuda-cupti-cu11 nvidia-cuda-nvrtc-cu11 nvidia-cuda-runtime-cu11

xuxuezhou, 2025-06-30:
> Hi, this is because cuda does not match the GPU. I used the following command to solve it. I hope it can help you! $ pip list | grep -i cuda nvidia-cuda-cupti-cu11 11.8.87 nvidia-cuda-cupti-cu12 [...]

Thanks for your reply! However, there’s no CUDA 11 installed on my system.

`$ pip list | grep -i cuda
nvidia-cuda-cupti-cu12   12.6.80
nvidia-cuda-nvrtc-cu12   12.6.77
nvidia-cuda-runtime-cu12 12.6.77
`

kellyguo11, 2025-07-07:
The blurry rendering issue should now be addressed in Isaac Sim 5.0.