isaac-sim/IsaacLab #2189: [Question] RL learning speed

jh-s5974, 2025-03-30:
I am using rsl-rl to do robot reinforcement learning. Is there any way to speed up the learning speed other than using a higher spec graphics card?

I am asking if there is a way to finish learning quickly because the learning time is long.

Thank you.

felipemohr, 2025-03-30:
The most obvious thing is to run the training in headless mode (but you're probably already doing that)

Depending on your application, and how accurate you need your physics to be, you can try reducing the update frequency. If your simulation is running at 200 Hz, training will occur faster than if it were running at 400 Hz, for example. But of course, you need to assess whether lowering the update frequency won't significantly impact your simulation

Something I've noticed makes a big difference is the decimation you use. Using a high decimation ends up slowing down the training, as you'll need a greater number of simulation updates to make a step in the training. For example, using a decimation of 8: 7 simulation updates will be needed, so that only on the eighth update is the data collected to train the policy. Of course, this will depend on how often you intend to implement your policy.

You can also try to reduce the number of envs in parallel (for example, train with only 1024 envs instead of 2048). This won't necessarily reduce the amount of time it takes for your training to finish, as you may need to train for more iterations to get the same result. But it's something I think is worth testing

Simplifying your robot's collisions reduces the amount of memory needed for training. I'm not sure if this also has a big impact on training speed, but it's something you can also try out

WangYCheng23, 2025-03-31:
Sampling can be time-consuming. To optimize, you can either increase the render interval or reduce the episode length, both of which can help save significant time

closed 2025-04-04 by RandomOakForest