isaac-sim/IsaacLab #2193: Unable to get correct force data form contact sensor

ItsGJun, 2025-03-31:
Hi, I'm trying to use contact sensor for the weight measure of water particels (this is the final purpose), and for the basic step, I'm checking if the contact sensor works well.

I wrote my code like right below.

import argparse

from isaaclab.app import AppLauncher

# create argparser
parser = argparse.ArgumentParser(description="Tutorial on spawning prims into the scene.")
parser.add_argument("--num_envs", type=int, default=3, help="Number of environments to spawn.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()
# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch

import isaaclab.sim as sim_utils
from isaaclab.assets import ArticulationCfg, AssetBaseCfg, RigidObjectCfg
from isaaclab.scene import InteractiveScene, InteractiveSceneCfg
from isaaclab.sensors import ContactSensorCfg
from isaaclab.utils import configclass

# Pre-defined configs
from isaaclab_assets.robots.franka import FRANKA_PANDA_CFG # isort: skip

class SceneCfg(InteractiveSceneCfg):

    # ground plane
    ground = AssetBaseCfg(prim_path="/World/GroundPlane", spawn=sim_utils.GroundPlaneCfg())

    # lights
    dome_light = AssetBaseCfg(prim_path="/World/Light", spawn=sim_utils.DomeLightCfg(intensity=3000.0, color=(0.75, 0.75, 0.75)))

    # articulation
    robot: ArticulationCfg = FRANKA_PANDA_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
    """ What is the usage of 'robot : ' ? """

    # # water & cup
    # usd_path_1 = "omniverse://localhost/Projects/Water_Env/Cup_Water_Flattened.usd"
    # cfg = sim_utils.UsdFileCfg(usd_path=usd_path_1)
    # cfg.func("/World/Objects/Cup_Water", cfg, translation=(0.0, 0.0, 1.05))

    # cube & contact sensor
    cube_1 = RigidObjectCfg(
        prim_path="{ENV_REGEX_NS}/Cube_1",
        spawn=sim_utils.CuboidCfg(
            size=(0.5, 0.5, 0.1),
            rigid_props=sim_utils.RigidBodyPropertiesCfg(),
            mass_props=sim_utils.MassPropertiesCfg(mass=100.0),
            collision_props=sim_utils.CollisionPropertiesCfg(),
            physics_material=sim_utils.RigidBodyMaterialCfg(static_friction=1.0),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0), metallic=0.2),
            activate_contact_sensors=True
        ),
        init_state=RigidObjectCfg.InitialStateCfg(pos=(0.5, 0.5, 0.05)),
    )

    cube_2 = RigidObjectCfg(
        prim_path="{ENV_REGEX_NS}/Cube_2",
        spawn=sim_utils.CuboidCfg(
            size=(0.2, 0.2, 0.1),
            rigid_props=sim_utils.RigidBodyPropertiesCfg(),
            mass_props=sim_utils.MassPropertiesCfg(mass=50.0),
            collision_props=sim_utils.CollisionPropertiesCfg(),
            physics_material=sim_utils.RigidBodyMaterialCfg(static_friction=1.0),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0), metallic=0.2),
            activate_contact_sensors=True
        ),
        init_state=RigidObjectCfg.InitialStateCfg(pos=(0.5, 0.5, 0.3)),
    )

    contact_forces_1 = ContactSensorCfg(
        prim_path="{ENV_REGEX_NS}/Cube_1",
        update_period=0.0,
        history_length=6,
        debug_vis=True,
        #filter_prim_paths_expr=["{ENV_REGEX_NS}/Cube_2"],
    )

def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):
    """Run the simulator"""
    # Define simulation stepping (not understood it yet..)
    sim_dt = sim.get_physics_dt()
    sim_time = 0.0
    count = 0

    # Simulate physics
    while simulation_app.is_running():
        # Reset
        if count % 500 == 0:
            # reset counter
            count = 0
            # reset the scene entities
            # root state
            # we offset the root state by the origin since the states are wrriten in simulation world frame
            # if this is not done, then the robots will be spawned at the (0,0,0) of the simulation world
            root_state = scene["robot"].data.default_root_state.clone()
            root_state[:, :3] += scene.env_origins
            scene["robot"].write_root_pose_to_sim(root_state[:, :7])
            scene["robot"].write_root_velocity_to_sim(root_state[:, 7:])
            # set joint positions with some noise
            joint_pos, joint_vel = (
                scene["robot"].data.default_joint_pos.clone(),
                scene["robot"].data.default_joint_vel.clone(),
            )
            joint_pos += torch.rand_like(joint_pos) * 0.1
            scene["robot"].write_joint_state_to_sim(joint_pos, joint_vel)
            # clear internal buffers
            scene.reset()
            print("[INFO]: Resetting robot state...")
        # Apply default actions to the robot
        # -- generate actions/commands
        targets = scene["robot"].data.default_joint_pos
        # -- apply action to the robot
        scene["robot"].set_joint_position_target(targets)
        # -- write data to sim
        scene.write_data_to_sim()
        # perform step
        sim.step()
        # update sim-time
        sim_time += sim_dt
        count += 1
        # update buffers
        scene.update(sim_dt)

        # print information from the sensors
        # # print(scene["contact_forces_1"].get_current_frame())
        # print("Received force matrix of: ", scene["contact_forces_1"].data.force_matrix_w)
        print("Received contact force of: ", scene["contact_forces_1"].data.net_forces_w)
        print("-------------------------------")

        """
        1. just one cube
        Received contact force of:  tensor([[[1.3181e-04, 0.0000e+00, 9.8100e+02]],

        [[1.3181e-04, 0.0000e+00, 9.8100e+02]],

        [[1.3181e-04, 0.0000e+00, 9.8100e+02]]], device='cuda:0')
        -------------------------------
        ->> right weights
        """

        """
        2. two cubes
        Received contact force of:  tensor([[[ 5.4774e-05, -1.6073e-04,  1.1281e+02]],

        [[-6.3875e-05, -3.0173e-05,  1.1282e+02]],

        [[-1.8863e-05, -2.6133e-04,  1.1282e+02]]], device='cuda:0')
        -------------------------------
        why printed like this?? three sensor environments are all the same!! ->> it's actually all the same on z ais

        ->>not right weights
        """

def main():
    """Main function."""

    # Initialize the simulation context
    sim_cfg = sim_utils.SimulationCfg(dt=0.005, device=args_cli.device)
    sim = sim_utils.SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view(eye=[3.5, 3.5, 3.5], target=[0.0, 0.0, 0.0])
    # design scene. scene instantiation
    scene_cfg = SceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
    scene = InteractiveScene(scene_cfg)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()

I checked if the contact sensor works well by two options.

First option was just placing one cube which weights 100kg. As a result, I could confirm it is right working cause on the z axis, the force was 100 * 9.81 = 981N

But the problem was the second option. I put the 50kg-weighted-cube on the first cube, and checked the force data. And the result was 1.1281e+02 and it is actually much smaller than 981N and weird result.. I don't know why it happens like this and I want to know why, and the way to correct this situation.

ItsGJun, 2025-03-31:
For more accurate situation explain, I post the picture of the simulation.

[image]

jtigue-bdai, 2025-04-01:
In the second option did you uncomment the `filter_prim_paths_expr=["{ENV_REGEX_NS}/Cube_2"],` line in the contact sensor definition?

ItsGJun, 2025-04-01:
Thank you for the reply!

<OPTION 2>
Yes, I checked it again right before with the simulation uncommented `filter_prim_paths_expr=["{ENV_REGEX_NS}/Cube_2"]`, and the force sensor result was still wrong. Here's the screenshot of it.

[image]

<OPTION 1>
And for the recheck, I simulated again with `filter_prim_paths_expr=["{ENV_REGEX_NS}/Cube_2"]` uncommented and also cube2 configuration uncommented. Here's the screenshot of it.

[image]

RandomOakForest, 2025-04-04:
Following up on this. Is the contact sensor you are using the Isaac Sim sensor or the PhysX sensor? Something is definitely off.

jtigue-bdai, 2025-04-04:
These are IsaacLab sensors. So they utilize the RigidContactView to access net_forces_w/force_matrix_w

RandomOakForest, 2025-04-16:
Following up, I'll move this to our _Discussions_ section. Let us know if you need further help.