isaac-sim/IsaacLab #2196: [Bug Report] Unable to include MotionViewer from external script

T-K-233, 2025-03-31:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Python cannot find `motion_loader` when trying to import MotionViewer from external package.

```bash
Traceback (most recent call last):
  File "/home/tk/Desktop/AIR/./scripts/poselib/view_motion.py", line 15, in <module>
    from isaaclab_tasks.direct.humanoid_amp.motions.motion_viewer import MotionViewer
  File "/home/tk/Documents/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid_amp/motions/motion_viewer.py", line 13, in <module>
    from motion_loader import MotionLoader
ModuleNotFoundError: No module named 'motion_loader'
```

### Steps to reproduce

Here's a minimal example program to reproduce the error.

```bash
from isaaclab.app import AppLauncher

# launch omniverse app
app_launcher = AppLauncher(headless=True)
simulation_app = app_launcher.app

"""Rest everything follows."""

from isaaclab_tasks.direct.humanoid_amp.motions.motion_viewer import MotionViewer
```

### System Info

Describe the characteristic of your environment:

- Commit: [b5fa0eb](https://github.com/isaac-sim/IsaacLab/commit/b5fa0eb031a2413c182eeb54fa3a9295e8fd867c)
- Isaac Sim Version: 4.5.0.0
- OS: Ubuntu 24.04
- GPU: RTX 4090
- CUDA: 12.6
- GPU Driver: 560.35.05

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-04-01:
Thank you for catching this. It seems line 13 in `motion_viewer.py` should be
```
from .motions import MotionLoader
```
We will update this. Thanks.

T-K-233, 2025-04-01:
Yes. Please see my PR #2195. Also I updated the `__init__.py` to allow importing the `MotionLoader` class from the .motion library level.

PR #2195 (2025-04-09): Fixes importing MotionViewer from external scripts
# Description

The `MotionViewer` class is not imported in `__init__.py`, preventing external scripts from using this class. This PR fixes this issue such that the MotionViewer class can be imported by external scripts.

The example usage would look like:

```python
from isaaclab.app import AppLauncher

# launch omniverse app
app_launcher = AppLauncher(headless=True)
simulation_app = app_launcher.app

"""Rest everything follows."""

from isaaclab_tasks.direct.humanoid_amp.motions import MotionViewer
```

Fixes #2196 

## Type of change

- Bug  [... cut]
commit 359cdc9c (2025-04-09): Fixes importing MotionViewer from external scripts (#2195)

# Description

The `MotionViewer` class is not imported in `__init__.py`, preventing
external scripts from using this class. This PR fixes this issue such
that the MotionViewer class can be imported by external scripts.

The example usage would look like:

```python
from isaaclab.app import AppLauncher

# launch omniverse app
app_launcher = AppLauncher(headless=True)
simulation_app = app_launcher.app

"""Rest everything follows."""

from isaaclab_tasks.direct.humanoid_amp.motions import MotionViewer
```

Fixes #2196 

## Type of chang [... cut]
closed 2025-04-09 by kellyguo11