isaac-sim/IsaacLab #2197: Light orientation

ChadMcintire, 2025-03-31:
If I start with a roll, pitch, and yaw of (0, 0, 90).

That means I want a Quaternion: (0.0, 0.0, 0.7071067811865475, 0.7071067811865476) for input if I am trying to set the orientation of a light using the following function:

cfg_light = sim_utils.CylinderLightCfg(intensity=3000.0, color=(1.0, 0.0, 0.0), length=5,)
cfg_light.func("/World/Light", cfg_light, translation=(0.0, 0.0, 1.5),  orientation=(0.0, 0.0, 0.7071067811865475, 0.7071067811865476))

In the simulation, this gives an orientation of  (-90, 0, 180)

-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
 
If I start with a roll, pitch, and yaw of (12,  270, 85).

That means I want a Quaternion: (-0.529591668978362, 0.46854312944350507, -0.5295916689783619, -0.46854312944350496),

In the app it shows an orientation of:
 
In the simulation, this gives an orientation of  (-90, 7, 90)

What am I missing?

RandomOakForest, 2025-04-01:
Thanks for posting this. Which version of Isaac Lab and Isaac Sim are you using? Thanks

Mayankm96, 2025-04-02:
The orientation is in wxyz format and not xyzw. Are you making sure this is correct?

closed 2025-04-02 by ChadMcintire