isaac-sim/IsaacLab #2199: [Bug Report] The calculation of rail_height is not suitable (row 400 in mesh_terrains.py)

AlexanderAbernathy, 2025-04-01:
### Describe the bug

For the terrain curriculum, the rail height calculation provides the maximum difficulty initially, which is unsuitable.
It should be : rail_height = cfg.rail_height_range[0] + difficulty * (cfg.rail_height_range[1] - cfg.rail_height_range[0])

The original version : rail_height = cfg.rail_height_range[1] - difficulty * (cfg.rail_height_range[1] - cfg.rail_height_range[0])

### Steps to reproduce

Multi-parallel RL for Anymal C in rail terrains.

-->

### Checklist

- [√ ] I have checked that there is no similar issue in the repo (**required**)
- [√] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-04-02:
Thank you for posting this. The team will review.

AntoineRichard, 2026-08-27:
Thanks @AlexanderAbernathy for the catch and the corrected one-liner, and sorry for the wait.

`source/isaaclab/isaaclab/terrains/trimesh/mesh_terrains.py` now reads `rail_height = cfg.rail_height_range[0] + difficulty * (cfg.rail_height_range[1] - cfg.rail_height_range[0])` — character-for-character your suggestion — so rails grow from the low end of the range as difficulty increases. Landed via https://github.com/isaac-sim/IsaacLab/pull/3254. Closing as resolved; please reopen if the terrain still looks off on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard