isaac-sim/IsaacLab #2210: [Question] Differentiate objects when using MultiAssetSpawnerCfg

luju97, 2025-04-01:
Hi,

I am using`MultiAssetSpawnerCfg` to spawn different objects in each environment.

I'm trying to train my robotic arm to pick up different types of objects, starting from the `FrankaCubeLiftEnvCfg` example.

How can I differentiate these objects during training? (I want to add this info to the observation space)
Is there a way to access information such as the usd path? I am unable to differentiate them since they are all under the same prim path.

Thank you in advance!

RandomOakForest, 2025-04-01:
Thanks for posting this. Check out [this tutorial on spawning multiple assets](https://isaac-sim.github.io/IsaacLab/main/source/how-to/multi_asset_spawning.html) and let us know if you need further help.

luju97, 2025-04-04:
@RandomOakForest I am using `MultiAssetSpawnerCfg`, like in the example:
```python
object: RigidObjectCfg = RigidObjectCfg(
    prim_path="/World/envs/env_.*/Object",
    spawn=sim_utils.MultiAssetSpawnerCfg(
        assets_cfg=[
            sim_utils.ConeCfg(
                radius=0.3,
                height=0.6,
                visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0), metallic=0.2),
            ),
            sim_utils.CuboidCfg(
                size=(0.3, 0.3, 0.3),
                visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(1.0, 0.0, 0.0), metallic=0.2),
            ),
            sim_utils.SphereCfg(
                radius=0.3,
                visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 0.0, 1.0), metallic=0.2),
            ),
        ],
        random_choice=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            solver_position_iteration_count=4, solver_velocity_iteration_count=0
        ),
        mass_props=sim_utils.MassPropertiesCfg(mass=1.0),
        collision_props=sim_utils.CollisionPropertiesCfg(),
    ),
    init_state=RigidObjectCfg.InitialStateCfg(pos=(0.0, 0.0, 2.0)),
)
```
I would like to know if there is a way to know whether the spawned object is a cone, a cube or a sphere (without using vision systems), just by getting this info from the simulator.

Thank you

RandomOakForest, 2025-04-04:
Is the `semantic_tags` attribute something you could use?

luju97, 2025-04-04:
@RandomOakForest If I use `semantic_tags` how can I access this info to add it to my observation space? E.g. something like a tensor of the tags for each environment.