isaac-sim/IsaacLab #2222: [Question] Package Import Warnings with isaaclab related modules

csyyysc, 2025-04-02:
### Question

I have followed [this guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html) to install Isaac Sim/Lab under miniconda virtual environment. And, the script running are normal after set up.

Yet, the import shows warning like `Import "isaaclab.sim" could not be resovled` for these similar code snippet.

```
"""Rest everything follows."""

import torch

import isaaclab.sim as sim_utils
from isaaclab.assets import ArticulationCfg, AssetBaseCfg
from isaaclab.scene import InteractiveScene, InteractiveSceneCfg
from isaaclab.sim import SimulationContext
from isaaclab.utils import configclass

##
# Pre-defined configs
##
from isaaclab_assets import CARTPOLE_CFG  # isort:skip
```

Furthermore, the `pip list` shows as follow:
```
Package                    Version        Editable project location
-------------------------- -------------- ----------------------------------------------------------------------------
isaaclab                   0.36.3         /home/me/Documents/Labs/isaaclabs-pip/IsaacLab/source/isaaclab
isaaclab_assets            0.2.1          /home/me/Documents/Labs/isaaclabs-pip/IsaacLab/source/isaaclab_assets
isaaclab_mimic             1.0.3          /home/me/Documents/Labs/isaaclabs-pip/IsaacLab/source/isaaclab_mimic
isaaclab_rl                0.1.3          /home/me/Documents/Labs/isaaclabs-pip/IsaacLab/source/isaaclab_rl
isaaclab_tasks             0.10.27        /home/me/Documents/Labs/isaaclabs-pip/IsaacLab/source/isaaclab_tasks
isaacsim                   4.5.0.0
isaacsim-app               4.5.0.0
isaacsim-asset             4.5.0.0
isaacsim-benchmark         4.5.0.0
isaacsim-code-editor       4.5.0.0
isaacsim-core              4.5.0.0
isaacsim-cortex            4.5.0.0
isaacsim-example           4.5.0.0
```

Is there anything that I missed out in the doc? That would be helpful to see typing hint when revising those snippets.

Mayankm96, 2025-04-02:
Does the script [here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#verifying-the-isaac-lab-installation) work fine?

Could you share the entire error and code snippet?

csyyysc, 2025-04-03:
Yes. The script works fine. Probably something is wrong with the pip pkg manager or IDE itself.  

[image]

csyyysc, 2025-04-09:
@Bardreamaster Thanks!! This works as expected! You save my days!

closed 2025-04-17 by RandomOakForest

chwenjun225, 2025-11-21:
> [@csyyysc](https://github.com/csyyysc) You can manually add `python.analysis.extraPaths` in your `.vscode/settings.json` which is in your current workspace directory. And remeber replace `${workspaceFolder}` with the IsaacLab directory. The `settings.json` might be like this:
> 
> {
>     // other configs.
> 
>     "python.analysis.extraPaths": [
>         "${workspaceFolder}/source/isaaclab_tasks",
>         "${workspaceFolder}/source/isaaclab_mimic",
>         "${workspaceFolder}/source/isaaclab",
>         "${workspaceFolder}/source/isaaclab_assets",
>         "${workspaceFolder}/source/isaaclab_rl"
>     ]
> }
> 
> By the way I have noticed that, in [.vscode/tools/setup_vscode.py](https://github.com/isaac-sim/IsaacLab/blob/main/.vscode/tools/setup_vscode.py#L67) the `isaaclab.sh` try to modify the `settings.json` under isaacsim installation directory but this file does not exist there. Even the config file exists, it looks it cannot help vscode to find the IsaacLab related packages.
> 
> So maybe only the maintainers know the details [@Mayankm96](https://github.com/Mayankm96) . Currently, if u want to have vscode import hints, just add the config manually.

Thank you for you solution, in my project it might be like this 

[image]

RESULT:
[image]