isaac-sim/IsaacLab #2227: [Question] Could not find configuration for the environment: 'Isaac-Velocity-Flat-H1-v0'.

eamooooon, 2025-04-02:
My IsaacLab version is 1.2.0
When I run this
`python source/standalone/workflows/rl_games/train.py --task Isaac-Velocity-Flat-G1-v0 --num_envs 16`
There is error like this
`ValueError: Could not find configuration for the environment: 'Isaac-Velocity-Flat-H1-v0'. Please check that the gym registry has the entry point: 'rl_games_cfg_entry_point'.`
How can I check my environment

RandomOakForest, 2025-04-02:
Thank you for posting this question. This may have been fixed in the latest [Isaac Lab release (v2.0.2)](https://github.com/isaac-sim/IsaacLab/tree/v2.0.2). Would it be possible for you to upgrade and try again? Also be sure your installation setup can be verified following the installation instructions.

eamooooon, 2025-04-03:
> Thank you for posting this question. This may have been fixed in the latest [Isaac Lab release (v2.0.2)](https://github.com/isaac-sim/IsaacLab/tree/v2.0.2). Would it be possible for you to upgrade [...]

unfortunately, i use viplanner extension which only tested with IsaacLab 1.2, so i cant update

RandomOakForest, 2025-04-23:
Following up on this, please start a new topic in our [_Discussions_](https://github.com/isaac-sim/IsaacLab/discussions) section if you need further help. I will close this issue for now. Thank you for your interest in Isaac Lab.

closed 2025-04-23 by RandomOakForest