isaac-sim/IsaacLab #2231: [Bug Report] convert_mjcf.py not working for G1

tylerlum, 2025-04-02:
I am trying to convert the G1 to USD from the unitree_ros repo (https://github.com/unitreerobotics/unitree_ros/tree/master/robots/g1_description)

When I run the following:

```
./isaaclab.sh -p scripts/tools/convert_mjcf.py \
  <path/to/unitree_ros>/robots/g1_description/g1_29dof_with_hand_rev_1_0.xml \ 
  source/isaaclab_assets/data/Robots/G1_mjcf_to_usd/g1_29dof_with_hand_rev_1_0.usd \      
  --import-sites \
  --make-instanceable
```

I get the following G1 (its hands look weird and its head is missing):
[image]

More detailed info of what is shown:
[image]

This is what I get when I visualize the same xml file in mujoco-python-viewer (https://github.com/rohanpsingh/mujoco-python-viewer) (it looks fine, so the xml should be fine):
[image]

tylerlum, 2025-04-02:
For reference, it worked for the urdf.

```
./isaaclab.sh -p scripts/tools/convert_urdf.py \
  <path/to/unitree_ros>/robots/g1_description/g1_29dof_with_hand_rev_1_0.urdf \
  source/isaaclab_assets/data/Robots/G1_urdf_to_usd/g1_29dof_with_hand_rev_1_0.usd \
  --merge-joints \
  --joint-stiffness 0.0 \
  --joint-damping 0.0 \
  --joint-target-type none
```

[image]

[image]

[image]

RandomOakForest, 2025-04-03:
Thank you for posting this. The team will review.

AntoineRichard, 2026-08-27:
@tylerlum thanks for reporting the broken G1 MJCF conversion, with the exact command line — that made this easy to place.

`convert_mjcf.py` has since been reworked onto the latest Isaac Sim MJCF importer (https://github.com/isaac-sim/IsaacLab/pull/5394 and https://github.com/isaac-sim/IsaacLab/pull/5889), with a different flag set: the `--import-sites` and `--make-instanceable` options in your command no longer exist, and the converter now emits asset-transformer output.

Closing as stale against the current tooling — please reopen with a fresh run on Isaac Lab 3.x if the G1 model still comes out with missing or wrong geometry.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard