isaac-sim/IsaacLab #2236: [Bug Report] RGB Image Not Correct When Setting Clip Values Smaller than 0.1

hzxie, 2025-04-03:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Here's the image when setting camera clip values equal to 0.1.

[image]

When changing this value to 0.05, the image becomes very strange.

[image]

### Steps to reproduce

Create a camera with the following config:

···
  cam_cfg = CameraCfg(
    ...
    spawn=sim_utils.PinholeCameraCfg(
        focal_length=2.3,
        focus_distance=400,
        horizontal_aperture=4.6,
        clipping_range=(0.01, 10000),
    ),
    ...
  )
···

### System Info

Describe the characteristic of your environment:

- Commit: f8cf6bad7
- Isaac Sim Version: 4.5.0-rc.36+release.19112.f59b3005.gl
- OS: Windows 11
- GPU: RTX 2080 Ti
- CUDA: 11.7.64
- GPU Driver: 566.24

### Additional context

N/A

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-04-03:
Thank you for catching this, the team will review.

Mayankm96, 2025-04-04:
The clip value defines the near plane of the camera image. It seems your camera is placed inside the mesh itself. When you have the near plane further away, you don't see the mesh.

More information on the planes is here: https://learnopengl.com/Getting-started/Coordinate-Systems

closed 2025-04-04 by hzxie