isaac-sim/IsaacLab #2238: [Bug Report] VS Code debugging issues for Python extension 2024.18.0 or newer

ozhanozen, 2025-04-03:
When running Python scripts using VScode launch configs inside the base IsaacLab Docker container (without a Conda env), I get errors such as "the `toml` module not being found". I believe other issues on the same problem were raised before, e.g., #1976 and #1500, which are left unresolved. I have debugged the problem a bit and found the following:

Apparently, starting with version 2024.18.0 of the VSCode Python extension, the new interpreter resolver (the “native locator”) now ignores shell‐script wrappers specified in the python.defaultInterpreterPath setting (see this [issue](https://github.com/microsoft/vscode-python/issues/24767)). This means IsaacLab Python environment is not setup correctly, resulting in the aforementioned errors.

Downgrading the VSCode Python extension version, using a conda environment, or adding the 
`"python.defaultInterpreterPath": "${workspaceFolder}/_isaac_sim/python.sh",
`
line to each of the launch.json entries seems to resolve the issue. Nevertheless, maybe there is a better way to do this.

### Steps to reproduce

Within the base IsaacLab container, use VSCode Python extension version 2024.18.0 or newer, no conda environment (i.e., use `_isaac_sim/python.sh`), and try to use the debugger to play any of the launch.json entries. You should get errors due to the environment not being set. Actually, you can already see in the terminal that `python.sh` is not executed, but rather the underlying Python executable directly.

### System Info

Describe the characteristic of your environment:

- Commit: bc7c9f5
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: 4090 RTX
- CUDA: 12.2
- GPU Driver: 535.129.03

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] VSCode Python debugger executing the correct bash/executable.
- [ ] If not, documentation in "Known Issues".

RandomOakForest, 2025-04-04:
Thank you for catching this and posting your detailed findings. We appreciate that. The team will review this. Thanks.

emirkin, 2025-07-21:
@RandomOakForest , any update on this issue?

EricPedley, 2025-07-30:
It seems like VSCode has changed the syntax for launch.json so now the line you would need to add is `"python": "${workspaceFolder}/_isaac_sim/python.sh"` (there is no "defaultInterpreterPath" property anymore).

EricPedley, 2026-01-28:
> I have created a new IsaacLabb project with `./isaaclab.sh --new` and am trying to debug my script with this config for the launch.json file
> 
> > {
> > "name": "Lander: Dreamer",
> > "type": "debugpy",
> > "request": "launch",
> > "python": "/workspace/isaaclab/_isaac_sim/kit/python/bin/python3",
> > "args" : ["--task", "Isaac-PlanetaryLander-Direct-6DOF-v0", "--num_envs", "8", "--enable_cameras"],
> > "program": "${workspaceFolder}/scripts/dreamer/train.py",
> > "cwd": "/workspace/SafeDreamer",
> > "console": "integratedTerminal",
> > "env": {
> > "ISAAC_SIM_PATH": "/workspace/isaaclab/_isaac_sim",
> > "CUDA_VISIBLE_DEVICES": "0"
> > }
> > },
> 
> But the debugger immediately stops and doesn't attach.
> 
> I can run the script from the project folder that was created like `python scripts/dreamer/train.py --task=Isaac-PlanetaryLander-Direct-6DOF-v0 --num_envs=8 --enable_cameras` and confirmed that works.
> 
> What am I missing from my configuration? [@RandomOakForest](https://github.com/RandomOakForest)

Try changing your python path to `/workspace/isaaclab/_isaac_sim/python.sh`. It runs some setup steps before running kit/python/bin/python3 on your script and those missing setup steps might be why you're having a problem.

george-nehma, 2026-01-29:
@EricPedley Unfortunately this is how I first had it and it still gave the error.

george-nehma, 2026-02-19:
@RandomOakForest Are there any updates for this issue, or how we might be able to run debugpy debugging sessions with the project format?

StafaH, 2026-09-22:
Fixed in #7940

closed 2026-09-22 by StafaH