isaac-sim/IsaacLab #2242: [Question] HUMANOID_28_CFG gets thrown into the air in tutorial create_scene.py

xindonglin99, 2025-04-03:
### Question

I'm not sure if this is a bug or it's because I did something wrong so I'll post it as a question first.

Context:
I'm running the tutorial file `scripts/tutorials/02_scene/create_scene.py` but if I replace the `CARTPOLE_CFG` with `HUMANOID_28_CFG`, I noticed that at each reset where I call `robot.write_root_pose_to_sim(root_state[:, :7])`, the humanoid will be thrown into the air (or bounce into the air, you can say). However, the `HUMANOID_CFG` model didn't experience the issue; it just fell on the ground.

The reason I'm doing this test is that I'm training an RL controller for another humanoid model but it can't learn anything because it's thrown into the air at the start of the episode and the episode just terminates. After further testing, I noticed that the `HUMANOID_28_CFG` has the same issue while `HUMANOID_CFG` doesn't. I tried to compare the config but the only things different are the stiffness and damping, which didn't seem to be the cause.

I'd appreciate it if you could look into this and provide some feedback.

AliBharwani, 2025-04-03:
Hey, I've been training the Humanoid28 config recently and ran into the same issue. The default pos of (0, 0, .8) is not high enough, so it spawns inside the ground and the physics engine tries to resolve this by pushing it up and out, resulting in the behavior you described. I changed it to (0, 0, 1.2) and that seems an appropriate starting position.

xindonglin99, 2025-04-03:
Thank you! That really helps!

> Hey, I've been training the Humanoid28 config recently and ran into the same issue. The default pos of (0, 0, .8) is not high enough, so it spawns inside the ground and the physics engine tries to [...]

closed 2025-04-03 by xindonglin99