isaac-sim/IsaacLab #2245: [Bug Report] Terrain curriculum only works with heading type Command

Nate711, 2025-04-04:
Since the curriculum is based on how far the robot walks from the env origin, it doesn't work when the velocity command is not the heading type. This is because such a command could have a angular velocity component that doesn't diminish to zero and instead makes the robot walk in a circle. In this case the robot could be doing great, but because it doesn't travel far from the origin, the curriculum difficulty will never go up.

https://github.com/isaac-sim/IsaacLab/blob/422cf358cebbf1fb7a55e59333d8245b8d46c46c/source/isaaclab_tasks/isaaclab_tasks/manager_based/locomotion/velocity/mdp/curriculums.py#L26

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] A warning is raised when this curriculum is used with a non-heading type command. 
- [ ] OR documentation is added for this failure mode. 
- [ ] OR the curriculum terrain function is updated to measure the velocity tracking error, not the L2 distance.

RandomOakForest, 2025-04-04:
Thanks for catching this. The team will review.

RandomOakForest, 2025-04-23:
Following up on this, the current implementation is designed to increase difficulty based on distance traveled from origin. However, it is possible to implement additional curricula based on other criteria. Please open a new topic in our _Discussions_ if you need further help. Thank you for your interest in Isaac Lab.

Nate711, 2025-04-27:
That's a good point but perhaps the documentation should be updated to let users know it only works with heading type commands.

Nate711, 2025-06-04:
@RandomOakForest This curriculum function is also broken if the velocity commands are not very high since the "move_up" condition is based on if the robot has escaped the current terrain grid cell. If the robot velocity target is say 0.25m/s, then the robot could be faithfully following this command but never escape the grid cell.

RandomOakForest, 2025-06-04:
Thanks for following up. These are very interesting observations. I think the implementation we have now may be somewhat limited, and likely needs clarifications in the docs of the supported behaviors and methods. I'll move this to a discussion for follow up for now. Thanks.