isaac-sim/IsaacLab #2253: gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist.

Radibi, 2025-04-05:
I am running the tutorial ：Modifying an existing Direct RL Environment.  but I get the error.
I am running it locally. 

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.230.02    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3080 Ti       | Yes: 0 |     | 12534   MB | 10de      | 0          |
|     |                                  |        |     |            | 2208      | 25df95dd.. |
|     |                                  |        |     |            | 65        |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-52-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Core(TM) i9-10900X CPU @ 3.70GHz
| Cores: 10 | Logical Cores: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 63981 | Free Memory: 57055
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
Traceback (most recent call last):
  File "/home/radibi/workspace/IsaacLab/scripts/reinforcement_learning/rl_games/train.py", line 184, in <module>
    main()
  File "/home/radibi/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 81, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
  File "/home/radibi/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 42, in register_task_to_hydra
    env_cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
  File "/home/radibi/anaconda3/envs/isaaclab/lib/python3.10/site-packages/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py", line 55, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.get(entry_point_key)
  File "/home/radibi/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 1029, in spec
    _check_version_exists(ns, name, version)
  File "/home/radibi/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 393, in _check_version_exists
    _check_name_exists(ns, name)
  File "/home/radibi/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 370, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist. Did you mean: `Isaac-Ant-Direct`?

*********************************************************************************************************************************************************

here is my __init__.py and h1_env.py
*************************************************************************************
`# Copyright (c) 2022-2025, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""
Humanoid locomotion environment.
"""

import gymnasium as gym

from . import agents
from .h1_env import H1Env, H1EnvCfg
##
# Register Gym environments.
##
gym.register(
    id="Isaac-Humanoid-Direct-v0",
    entry_point=f"{__name__}.humanoid_env:HumanoidEnv",
    disable_env_checker=True,
    kwargs={
        "env_cfg_entry_point": f"{__name__}.humanoid_env:HumanoidEnvCfg",
        "rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",
        "rsl_rl_cfg_entry_point": f"{agents.__name__}.rsl_rl_ppo_cfg:HumanoidPPORunnerCfg",
        "skrl_cfg_entry_point": f"{agents.__name__}:skrl_ppo_cfg.yaml",
    },
)

gym.register(
    id="Isaac-H1-Direct-v0",
    entry_point="isaaclab_tasks.direct.humanoid:H1Env",
    disable_env_checker=True,
    kwargs={
        "env_cfg_entry_point": H1EnvCfg,
        "rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",
        "rsl_rl_cfg_entry_point": f"{agents.__name__}.rsl_rl_ppo_cfg:HumanoidPPORunnerCfg",
        "skrl_cfg_entry_point": f"{agents.__name__}:skrl_ppo_cfg.yaml",
    },
)`

**********************************************
# Copyright (c) 2022-2025, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from __future__ import annotations

from isaaclab_assets import H1_CFG

import isaaclab.sim as sim_utils
from isaaclab.assets import ArticulationCfg
from isaaclab.envs import DirectRLEnvCfg
from isaaclab.scene import InteractiveSceneCfg
from isaaclab.sim import SimulationCfg
from isaaclab.terrains import TerrainImporterCfg
from isaaclab.utils import configclass

from isaaclab_tasks.direct.locomotion.locomotion_env import LocomotionEnv

@configclass
class H1EnvCfg(DirectRLEnvCfg):
    # env
    episode_length_s = 15.0
    decimation = 2
    action_scale = 1.0
    action_space = 21
    observation_space = 75
    state_space = 0

    # simulation
    sim: SimulationCfg = SimulationCfg(dt=1 / 120, render_interval=decimation)
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="average",
            restitution_combine_mode="average",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
        debug_vis=False,
    )

    # scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=4096, env_spacing=4.0, replicate_physics=True)

    # robot
    robot: ArticulationCfg = H1_CFG.replace(prim_path="/World/envs/env_.*/Robot")
    joint_gears: list = [
        50.0,  # left_hip_yaw
        50.0,  # right_hip_yaw
        50.0,  # torso
        50.0,  # left_hip_roll
        50.0,  # right_hip_roll
        50.0,  # left_shoulder_pitch
        50.0,  # right_shoulder_pitch
        50.0,  # left_hip_pitch
        50.0,  # right_hip_pitch
        50.0,  # left_shoulder_roll
        50.0,  # right_shoulder_roll
        50.0,  # left_knee
        50.0,  # right_knee
        50.0,  # left_shoulder_yaw
        50.0,  # right_shoulder_yaw
        50.0,  # left_ankle
        50.0,  # right_ankle
        50.0,  # left_elbow
        50.0,  # right_elbow
    ]

    heading_weight: float = 0.5
    up_weight: float = 0.1

    energy_cost_scale: float = 0.05
    actions_cost_scale: float = 0.01
    alive_reward_scale: float = 2.0
    dof_vel_scale: float = 0.1

    death_cost: float = -1.0
    termination_height: float = 0.8

    angular_velocity_scale: float = 0.25
    contact_force_scale: float = 0.01
    action_space = 19
    observation_space = 69

class H1Env(LocomotionEnv):
    cfg: H1EnvCfg

    def __init__(self, cfg: H1EnvCfg, render_mode: str | None = None, **kwargs):
        super().__init__(cfg, render_mode, **kwargs)

yaoshiluo, 2025-04-06:
Maybe your environment ID should be Isaac-Humanoid-Direct-v0 instead of Isaac-H1-Direct

yaoshiluo, 2025-04-06:
> Maybe your environment ID should be Isaac-Humanoid-Direct-v0 instead of Isaac-H1-Direct

or it should be Isaac-H1-Direct-v0

Radibi, 2025-04-06:
> Maybe your environment ID should be Isaac-Humanoid-Direct-v0 instead of Isaac-H1-Direct

I follow this [https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/modify_direct_rl_env.html](url). It is "Modifying an existing Direct RL Environment".  ---isaaclab 4.5
 In the tutorial it add a gym.register in the source/isaaclab_tasks/isaaclab_tasks/direct/humanoid/__init__.py and use the ID：Isaac-H1-Direct-v0. And run `./isaaclab.sh -p scripts/reinforcement_learning/rl_games/train.py --task Isaac-H1-Direct-v0 --headless`.  

If i run `./isaaclab.sh -p scripts/reinforcement_learning/rl_games/train.py --task Isaac-Humanoid-Direct-v0 --headless` . I will get the original humanoid env.

Radibi, 2025-04-06:
> > 也许你的环境 ID 应该是 Isaac-Humanoid-Direct-v0 而不是 Isaac-H1-Direct
> 
> 或者应该是 Isaac-H1-Direct-v0

closed 2025-04-06 by Radibi